LiberTEM/LiberTEM #714: Cluster connection form: improve UX after error

sk1p, 2020-04-12:
See comment in #699:

> There is still a small user experience issue, which could be fixed in a follow-up PR: after a failed connection attempt, the form goes back to its original state (both the type drop-down and the connection URL). If a user runs into this bug, they probably want to re-try with similar parameters, for example if they had a typo in the URL, or the remote cluster was not running yet. So it would be really nice if the form re-opened with the original parameters.

twentyse7en, 2020-04-17:
Can I apply the following approach, save the current URL and cluster type when the form is submitting. When cluster connection fails, re-open form with saved values.

sk1p, 2020-04-17:
Sounds like it could work - should be put into the localstorage together with the rest of the config