LiberTEM/LiberTEM #725: Size of task results exchanged with Dask cluster suspected to be 16x inflated

uellue, 2020-04-16:
LiberTEM is unusably slow when using a remote Dask cluster over a typical broadband internet connection of about 1 MB/s, with task results coming in only every few seconds. The system monitor shows maxed-out network traffic while an analysis is running and drops to almost zero background load when it finishes.

This is in contradiction with the expected size of task results: On 256x256 frames, a task result from `SumUDF` should have a size of  256 kB. That means it should manage about four updates per second. Instead, it manages only one update every four seconds.

Setup: LiberTEM master, Python 3.6, LiberTEM server on Windows, Dask cluster on Linux (moellenstedt).

Unfortunately, I didn't manage to capture the network traffic because sniffing localhost on Windows  is difficult.

@sk1p can you reproduce the issue with LiberTEM server on Linux?

sk1p, 2020-04-16:
> can you reproduce the issue with LiberTEM server on Linux?

Thanks for the report, i'll have a look today.

sk1p, 2020-04-16:
Did this happen with all `DataSet`s, or with one in particular?

sk1p, 2020-04-16:
I could reproduce this with a MIB dataset. I think I know the cause and I'll work on a proper fix.

uellue, 2020-04-16:
Great that you could reproduce it! I am tentatively including it in 0.5 since you are working on a fix already. :-)

closed 2020-04-16 by sk1p

sk1p, 2020-04-16:
This is now fixed, note that for very large detectors, the result transfer is still the bottleneck (related to #134 and #242)

PR #729 (2020-04-16): Reduce UDF result overhead
Fixes #725 

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [ ] I have added/updated test cases