LiberTEM/LiberTEM #726: Refactor: info part of dataset detect results should not be part of the dataset params type

sk1p, 2020-04-16:
See discussion in the commits of #700 - instead of keeping [a "blacklist" of info parameters](https://github.com/LiberTEM/LiberTEM/blob/5cfd119e6eea084011f08278d71aa90f05666939/client/src/dataset/components/DatasetParams.tsx#L21), the ["info" part (dataset_paths here)](https://github.com/LiberTEM/LiberTEM/blob/5cfd119e6eea084011f08278d71aa90f05666939/client/src/messages.ts#L105) should be properly separated from the rest of [the `DatasetFormParams` type](https://github.com/LiberTEM/LiberTEM/blob/5cfd119e6eea084011f08278d71aa90f05666939/client/src/messages.ts#L152)

AnandBaburajan, 2020-04-17:
Note to self: Fix https://github.com/LiberTEM/LiberTEM/pull/732 too

sk1p, 2020-06-03:
See also: #793

PR #793 (2020-12-14): sync_offset, reshape datasets and get coords for UDFs
## Google Summer of Code '20 project

Fixes #256, fixes #553 and fixes #726 and

* The Python API and GUI now allows specifying `nav_shape` and `sig_shape` parameters to set a different shape
* `scan_size` and `detector_size` parameters are deprecated in favor of `nav_shape` and `sig_shape` respectively
* All datasets handle missing data gracefully
* The Python API and GUI now allow specifying a `sync_offset` to handle synchronization/acquisition problems
* Users can now access the coordinates of a tile/partition slice through UDFMeta.coord [... cut]
closed 2020-12-14 by sk1p