LiberTEM/LiberTEM #741: Glitch in file opening dialogue

uellue, 2020-04-22:
How to reproduce in the GUI, in this case mixing EMPAD XML and BLO:

* Click "+Browse"
* Select a file to get the file loading dialogue
* Click "+Browse" again while the previous dialogue is still open
* Select another file
* Click "Load Dataset"

Observed behavior:
* Only one file loading dialogue is visible even if a second file was selected
* The parameters from both file types seem to get mixed up when the dialogue is submitted.

Expected behavior: Two loading dialogues in parallel that work independently and don't mix, or no further browsing possible before loading is concluded.

AnandBaburajan, 2020-04-22:
@uellue Can I look into and try to solve this?

sk1p, 2020-04-23:
> Can I look into and try to solve this?

Sure, go ahead!

PR #752 (2020-04-27): Fix parallel file browsing glitch
Fixes #741 
> No further browsing possible before loading is concluded

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [x] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html?#buil [... cut]
closed 2020-04-27 by sk1p