LiberTEM/LiberTEM #743: Add logging level argument to cli

sk1p, 2020-04-22:
To allow users to quickly debug something, the logging level should be configurable from the cli.

Iamshankhadeep, 2020-04-26:
is it done?

sk1p, 2020-04-27:
> is it done?

No, the commits referencing this issue statically set the logging level to `INFO`. This issue is about allowing the user to specify the logging level from the CLI.

sayandip18, 2020-04-27:
> 
> This issue is about allowing the user to specify the logging level from the CLI.

By "level" do you mean `DEBUG`, `INFO`, `WARNING`, etc?

sk1p, 2020-04-27:
> By "level" do you mean `DEBUG`, `INFO`, `WARNING`, etc?

Exactly.

sayandip18, 2020-04-27:
How can the user specify these? Aren't they supposed to be in the code?

sk1p, 2020-04-27:
Log levels appear in different places. One is in code - some function decides to call `someLogger.debug(...)`, and a message with level DEBUG is logged. Then, each logger can be configured to only log specific levels, for example, the code could contain `someLogger.debug(..)` statements, but the logger will only pass them on to the handler if the logger level is set appropriately. Then, handlers can specify the threshold they want to log - if a handler is set to INFO and gets a log message with level DEBUG, it will not handle that message. See also [the official documentation](https://docs.python.org/3/library/logging.html) for details.

The idea of this issue is to allow the user to configure *the logging handler*. By default, only log messages with level INFO should be logged, but if the user hits a problem, they should be able to switch over to DEBUG logging and also get log messages with level DEBUG. Or they could configure the handler level to WARNING or ERROR, and only get more important messages.

PR #758 (2020-04-29): Cmd line argument to set log level
Logging level can be configured from cmd line, default INFO.
fix #743 

## Contributor Checklist:

* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
closed 2020-04-29 by sk1p