LiberTEM/LiberTEM #754: Remove tileshape and enable_direct parameters where they still exist

sk1p, 2020-04-27:
Cleanup work from #706.

`tileshape` should be removed/ignored without replacement.

As for `enable_direct`, we can still support the parameter for a while after #716 is done, but it should be moved to a "select i/o engine" kind of parameter (#753), such that it is available for (almost) all formats.

uellue, 2023-04-20:
Hm, seems to be implemented in https://github.com/LiberTEM/LiberTEM/blob/994141ed0952b7d73a530dccdb492b178db9880a/src/libertem/io/dataset/raw.py#L102-L110

Not sure if we close here and leave as-is, or make this a proper deprecation in 0.12? Bumping to next release for evaluation.

sk1p, 2023-07-26:
I think we can close this - we don't really lose too much by keeping the backwards compatibility.

closed 2023-07-26 by sk1p