LiberTEM/LiberTEM #756: Add test cases for reshaped_view

sk1p, 2020-04-27:
Extracted from #706: Add test cases for flattening the signal dimension of tiles using `libertem.common.buffers.reshaped_view()` #739

sayandip18, 2020-04-27:
What about [this](https://github.com/LiberTEM/LiberTEM/blob/b8f3c6102142d9fceac1efa790d9d6686f55bce9/tests/common/test_bufferwrapper.py#L57)?

sk1p, 2020-05-04:
> What about [this](https://github.com/LiberTEM/LiberTEM/blob/b8f3c6102142d9fceac1efa790d9d6686f55bce9/tests/common/test_bufferwrapper.py#L57)?

I think this issue was about a more specific test case, @uellue should know details.

uellue, 2020-05-04:
@sayandip18 @sk1p This is a test for the function itself, to make sure that it throws an error if the received data can't be flattened without a copy. Test cases for the rest of the system are still missing:

* Make sure only contiguous views are delivered, even if the signal dimension is sliced unfavourably
* Make sure the system throws an error if the slices in the cache overlap
* Make sure the system throws an error if the underlying data is accessed the wrong way while a cached copy exists
* Make sure the contiguous copies are merged back correctly

This is only possible with the new infrastructure for tiled `MemoryDataSet` of #706, while the fix with contiguous copies had to go into 0.5.0 to make `StdDevUDF` work correctly.

uellue, 2020-06-17:
Refs #814, discussion with @sk1p: Remove test of disjointness from production code and integrate it with the contiguous tests -- call run_for_partition() and check afterwards that the cache keys are disjoint.

PR #819 (2020-06-25): Fix #756: Fix part of #814; basic benchmarking
Using https://pytest-benchmark.readthedocs.io/en/latest/ to pinpoint the performance issue

Closes #756 
Refs #198 

This shows significant improvement when getting contiguous copies of a `kind="sig"`buffer, but doesn't seem to fix the main part of #814, unfortunately.

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for [... cut]
commit 35db71b9 (2020-06-25): Test contiguous copies; Alleviate performance regression; Basic benchmarking

Closes #756 : Test for contiguous copies working correctly

Refs #814 : Fix parts of the performance regression

Refs #198 : Introduce first benchmark to reproduce parts of the performance regression in #814

Include basic infrastructure to use https://pytest-benchmark.readthedocs.io/en/latest/.

It fulfills many requirements in #198. Most notably, it allows to integrate benchmarks
seamlessly in unit tests. Collecting data allows to build a track record. The data is in a format that
makes developing a custom reportin [... cut]
closed 2020-06-25 by uellue