LiberTEM/LiberTEM #791: Dataset dimension squeezed out

uellue, 2020-05-20:
How to reproduce:

Process data with nav shape (1,), possibly with aux data

Observed behavior:
```
c:\users\weber\documents\libertem\libertem\src\libertem\common\buffers.py in set_buffer(self, buf, is_global)
    503             extra = (1,)
    504         shape = (-1,) + extra
--> 505         self._data = buf.reshape(shape).squeeze()
    506         self._data_coords_global = is_global
    507 

ValueError: cannot reshape array of size 2 into shape (49,2)
```
squeeze() is likely squeezing out the ds dimension.

sk1p, 2020-05-20:
That's a nice edge case, good find!

sayandip18, 2020-07-29:
> Process data with nav shape (1,), possibly with aux data

Is that the `extra_shape` in buffer declaration that is supposed to be (1,)?

PR #902 (2020-11-18): Fix AUX data failure with single frame partitions
If a partition consists of a single "sig" item, aux data shapes don't match. The singular dimension is "swallowed" somehow, apparently. I tried to find out where, but didn't get far.

This is mostly an issue for testing since normally partitions contain more than one sig element. For memory partitions, the "num_partitions" parameter can be set in such a way that no partitions with singular nav shape are created as a work-around.

Fixes #791 

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://g [... cut]
closed 2020-11-18 by sk1p