LiberTEM/LiberTEM #796: GUI: ROI result is black

jhgee, 2020-05-29:
When selecting a rectangle ROI which is too small the result stays black.

## works:
[image]
## doesn't work:
[image]
## works:
[image]
## doesn't work:
[image]

jhgee, 2020-05-29:
Same is true for disk ROIs

## works:
[image]

## doesn't work:
[image]

sk1p, 2020-05-29:
Is this using the master branch, or @AnandBaburajan's PR #793? Is this only happening with standard deviation, or also with average?

jhgee, 2020-05-29:
@sk1p this happened indeed using the PR #793 revision https://github.com/LiberTEM/LiberTEM/pull/793/commits/391e8d498d417cfb92f12bb03e12acf8449533d3
It seems also depend on the dataset I loaded. With the current dataset I loaded I can't reproduce the issue but also got the zero devision error.
I will test average once I can reproduce the error again.

jhgee, 2020-05-29:
@sk1p With the dataset from above also average leads to the same result. 
I thought about testing on the master branch version but this will not work, at leas with this unsynchronized dataset yet. I could try it with a synchronized one.

sk1p, 2020-05-29:
> I could try it with a synchronized one.

That would be good - my guess is that region of interest handling also needs changes in #793

jhgee, 2020-05-29:
I tested another dataset where even stuff like this works
[image]

jhgee, 2020-05-29:
seems not to be a problem in general
[image]

AnandBaburajan, 2020-05-29:
> @AnandBaburajan could you have a look if this is related to your PR? Basically, making sure that regions of interest still work with your changes. That's also something that should get its own test case.

Yup, I'll look into this.

AnandBaburajan, 2020-06-01:
> @AnandBaburajan could you have a look if this is related to your PR? Basically, making sure that regions of interest still work with your changes. That's also something that should get its own test case.

Yup, this was related to my PR. I've fixed it and I'll add tests after the implementation is complete.

> When selecting a rectangle ROI which is too small the result stays black.

This has been fixed.

> sometimes this is accompanied by:
> [image]

I never got this error though.

sk1p, 2020-06-02:
> I never got this error though.

Did you test with standard deviation, as in the screenshot? If you decide to retry this and do manage to reproduce it, I would be interested in a traceback (should be on the console output) , as this could possibly be an unhandled boundary condition somewhere.

AnandBaburajan, 2020-06-02:
> Did you test with standard deviation, as in the screenshot? If you decide to retry this and do manage to reproduce it, I would be interested in a traceback (should be on the console output) , as this could possibly be an unhandled boundary condition somewhere.

Sorry! I only tried the average mode. I just tried ROIs with standard deviation both on mine and the master branch and there's no result when the ROI is really small.

Regarding my PR, many rows of frames were giving no results in the average mode and that's been fixed.

But I still couldn't reproduce the `division by zero` error.

sk1p, 2020-06-02:
> there's no result when the ROI is really small.

If the ROI only spans a single frame, this is expected, as the standard deviation of a single value should be zero, IIRC.

> Regarding my PR, many rows of frames were giving no results in the average mode and that's been fixed.

:+1: