LiberTEM/LiberTEM #798: Ensure jitted functions can be cached

sk1p, 2020-06-02:
If we can cache them, we can get rid of of some nasty latencies introduced by #706, which is important for a smooth user experience.

sk1p, 2020-09-21:
As an update, I filed https://github.com/numba/numba/issues/6264

I also checked the potential savings using a patched numba that can cache functions captured in a closure. Measured via the GUI on an 8GiB MIB dataset, LiberTEM-stable (v0.5.0) takes about 3.8s to compute the initial sum of all frames, where LiberTEM-master+patches takes 4.6s. I'll retry on a larger data set in a bit, there the throughput improvements should outweigh the initial additional latency (by a noticeable margin)

Maybe the remaining latency can be hidden by heating up the cache directly after loading a data set, as we discussed before - it could possibly also be reduced in the cache miss case, as it is enough to run one compilation job per host to populate the cache, instead of having multiple redundant compilations (for each core) competing for resources.

uellue, 2020-10-08:
In case numba/numba#6284 is not included in the next Numba release, we could perhaps implement a workaround to get the 0.6 release out in reasonable time.

We only need this caching in a well-defined subset of LiberTEM that has a well-defined interface with user code: the I/O routines.  Furthermore, the caching in Numba is implemented in a central place and your patch seems to work. We could implement a context manager that monkey-patches the Numba cache with your patch *only* when LiberTEM enters I/O routines. That means user code will not be affected by the monkey patching.

Heating up the cache directly after loading sounds like a good idea! It felt more acceptable to wait when opening instead of waiting for the results to come in after adding an analysis or clicking "Apply".

sk1p, 2020-10-12:
> We only need this caching in a well-defined subset of LiberTEM that has a well-defined interface with user code: the I/O routines. Furthermore, the caching in Numba is implemented in a central place [...]

This monkey-patched hack would need to make sure that the cache key includes some unique element that makes sure the cache is busted once the hack is removed. I'll have a look into this.

> Heating up the cache directly after loading sounds like a good idea! It felt more acceptable to wait when opening instead of waiting for the results to come in after adding an analysis or clicking "Apply".

This is now done in my branch - I could clean that up to a mergable status, only including the warmup and enabling of jit caching,  which should "work" even without the numba changes. Then I can work on a monkey-patched hack in a follow-up. Does that sound good?

uellue, 2020-12-15:
Opening a small RAW dataset now takes 1,5 minutes on my system after #886 was merged, which feels like an eternity while watching a spinner. Reopening here since #886 is only part of the solution.

uellue, 2020-12-15:
For reference, the first run of a point analysis took about 30 s on the same system before. That means without actual caching it seems that #886 introduced some inefficiencies.

sk1p, 2020-12-15:
> That means without actual caching it seems that #886 introduced some inefficiencies.

I'll have a look later. Thanks for the report! Is that with the latest numba version?

uellue, 2020-12-15:
> Is that with the latest numba version?

This was 0.51.2. Retesting now with 0.52.

uellue, 2020-12-15:
With 0.52 it is 1:08 min to open a RAW dataset for the second run, i.e. with everything warmed up.

uellue, 2020-12-21:
Repurposed to remind us of removing the hack of the Numba cache #931 as soon as Numba supports caching these functions natively.

uellue, 2025-03-27:
https://github.com/numba/numba/issues/6264 is still open and the caching issue still seems to persist when trying out the reproducer with current versions.

PR #886 (2020-12-15): JIT caching updates
Fixes #798 once numba properly caches functions with w/ closures over other functions. Independent from this, we can decide if we want to add a work-around to make these functions cacheable. 

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [x] I have added/ [... cut]