LiberTEM/LiberTEM #815: Enable GPU support for holography UDF

sk1p, 2020-06-16:
As of #760, the GPU support in the holography UDF is not completely finished, as it fails under not-yet-debugged circumstances. As such, it will be disabled for the 0.6 release, and we will work on debugging it for 0.7

sk1p, 2020-06-16:
It fails on my system with `cupy.cuda.cufft.CuFFTError: CUFFT_EXEC_FAILED`:

```
src/libertem/api.py:707: in run_udf
    results = UDFRunner([udf]).run_for_dataset(
src/libertem/udf/base.py:1113: in run_for_dataset
    for part_results, task in executor.run_tasks(tasks, cancel_id):
src/libertem/executor/inline.py:23: in run_tasks
    result = task()
src/libertem/udf/base.py:847: in __call__
    return UDFRunner(self._udfs).run_for_partition(
src/libertem/udf/base.py:1063: in run_for_partition
    self._run_udfs(numpy_udfs, cupy_udfs, partition, tiling_scheme, roi, dtype)
src/libertem/udf/base.py:995: in _run_udfs
    run_tile(cupy_udfs, tile, device_tile)
src/libertem/udf/base.py:973: in run_tile
    udf.process_frame(frame)
src/libertem/udf/holography.py:198: in process_frame
    fft_frame = self.xp.fft.fft2(frame) / np.prod(frame_size)
../../.virtualenvs/libertem/lib/python3.8/site-packages/cupy/fft/fft.py:532: in fft2
    return func(a, s, axes, norm, cufft.CUFFT_FORWARD)
../../.virtualenvs/libertem/lib/python3.8/site-packages/cupy/fft/fft.py:417: in _fftn
    a = _exec_fftn(a, direction, value_type, norm=norm, axes=axes,
../../.virtualenvs/libertem/lib/python3.8/site-packages/cupy/fft/fft.py:365: in _exec_fftn
    plan.fft(a, out, direction)
cupy/cuda/cufft.pyx:298: in cupy.cuda.cufft.PlanNd.fft
    ???
cupy/cuda/cufft.pyx:346: in cupy.cuda.cufft.execZ2Z
    ???
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

>   ???
E   cupy.cuda.cufft.CuFFTError: CUFFT_EXEC_FAILED

cupy/cuda/cufft.pyx:98: CuFFTError
```

Once the error is hit, the CUDA context is in a strange state and can't execute further cuFFT or cuBLAS calls, for example:

```
(Pdb) import cupy
(Pdb) aaa = cupy.ones((1024, 1024))
(Pdb) bbb = cupy.ones((1024, 1024))
(Pdb) aaa @ bbb
*** cupy.cuda.cublas.CUBLASError: CUBLAS_STATUS_EXECUTION_FAILED
(Pdb) 
```

uellue, 2020-06-16:
Previous discussion in https://github.com/LiberTEM/LiberTEM/pull/760#discussion_r438282154

uellue, 2025-03-27:
Closing here since now taken care of in https://github.com/LiberTEM/LiberTEM-holo

closed 2025-03-27 by uellue