LiberTEM/LiberTEM #818: Correct function does not fix bad pixel

bangunarya, 2020-06-18:
I recently tried to implement correction of bad pixel using correct function
```sig_shape = (256,256)```
```x =np.array([(240,174,206,143), (157,228,242,41)]).astype(np.int64)```
```data_new = correct(data, dark_image = None, gain_map = None, excluded_pixels = x, inplace = False, sig_shape = sig_shape)```

However, when summing all frames it seems the correct function does not change the specific bad pixel from the surrounding pixel. Is there something wrong with the implementation?
[image]

uellue, 2020-06-18:
Hi @bangunarya Thx for the report! Can you perhaps include how you generated data and how you plotted it?

sk1p, 2020-06-18:
I tried this with synthetic data, and it appears to work fine:

[image]

uellue, 2020-06-18:
In that case we should have a look at the entire notebook to see where the issue is.

uellue, 2020-06-18:
It seems to have done *something* since the leftmost and rightmost pixel went from bright to dark.

bangunarya, 2020-06-19:
> Hi @bangunarya Thx for the report! Can you perhaps include how you generated data and how you plotted it?

bangunarya, 2020-06-19:
> Hi @bangunarya Thx for the report! Can you perhaps include how you generated data and how you plotted it?
Hi Dieter, here I attached the code. 
[image]

uellue, 2020-06-19:
Thank you! I'll try it out next week and see what happens there.

uellue, 2020-06-23:
@bangunarya Thx again for reporting! I could reproduce it, it was an issue with handling integer input data. PR #831 fixes this on my system.

PR #831 (2020-07-06): Make sure correction handles integer input data correctly
Closes #818 

Correction only makes sense for floating point data.

* Inplace: Throws TypeError for non-float data
* Copy: Cast to matching floating point type.

@sk1p How is it handled in the dataset corrections code? Do we make sure it is float when correction is turned on?

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/sou [... cut]
commit 1d1c9109 (2020-07-06): Only correct floating point data

Closes #818
closed 2020-07-06 by sk1p