LiberTEM/LiberTEM #833: numpy deprecation warning in corrections test

sk1p, 2020-06-24:
```
  /Users/travis/build/LiberTEM/LiberTEM/tests/corrections/test_detector.py:295: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray
    indices = np.array([np.random.randint(low=0, high=s, size=s//2) for s in (20, ) + sig_dims])
```

sayandip18, 2020-06-25:
What are "corrections"?

sk1p, 2020-06-25:
Basically, sometimes there are artifacts in the raw data that can be corrected, for example different responses of single pixels. We recently added support for three of these corrections, namely dark frame subtraction, gain maps and defect pixel patching.

sayandip18, 2020-06-28:
I think adding `dtype=object` will solve this. All Travis CI tests will be successful.

sk1p, 2020-07-20:
The question is, is that the right thing to do in that case? If I read it correctly, the array in question is not even a ragged array, or is it?

sayandip18, 2020-07-20:
The latest [commit](https://github.com/LiberTEM/LiberTEM/commit/117f4adcce5ff685a6f455821d54311d29f11d93) has pushed line 295 to line [350](https://github.com/LiberTEM/LiberTEM/blob/1b372535e303a5fa9ae715033ec5f151626cc099/tests/corrections/test_detector.py#L350).

Here's how it goes:

In [`test_mark_correction_sparse()`](https://github.com/LiberTEM/LiberTEM/blob/1b372535e303a5fa9ae715033ec5f151626cc099/tests/corrections/test_detector.py#L323), the variable `num_sig_dims` in line [327](https://github.com/LiberTEM/LiberTEM/blob/1b372535e303a5fa9ae715033ec5f151626cc099/tests/corrections/test_detector.py#L327) chooses between 2 and 3 at random. `sig_dims` in line [330](https://github.com/LiberTEM/LiberTEM/blob/1b372535e303a5fa9ae715033ec5f151626cc099/tests/corrections/test_detector.py#L330) is a random tuple of size `num_sig_dims` and values between 8 and 16. Now, in line 350, `(20,) + sig_dims` gives a new tuple, the first element of which is 20 and the rest are same as `sig_dims`. As `s` iterates through each element of this tuple, new ndarrays are returned by the `randint()` function which are of size `s//2` which is same as `floor(s/2)`

 `np.random.randint()` in `sig_dims` ensures that the array `indices` is a ragged one

sk1p, 2020-07-20:
Ahh, I see, I overlooked the `size=s//2`! One could also just make `indices` a list-of-arrays instead (remove the `np.array` call), as no numpy functions are called on that anyways...

PR #853 (2020-07-24): Fix some warnings in test code
Closes #833

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
commit 7eb53861 (2020-07-24): Fix some warnings in test code

Closes #833
closed 2020-07-24 by uellue