LiberTEM/LiberTEM #835: Input fields in FFT analysis don't work in MS edge

uellue, 2020-06-29:
How to reproduce:

* LiberTEM release version 0.5.0
* Edge version: 16.16299.547.0

* Open a dataset in the GUI
* add an FFT analysis
* Try to change parameters

What happens:

* The field gets blanked
* One can't enter numbers

What I expect:

* I can modify the value in the field.

uellue, 2020-06-29:
Actually, many GUI features including keyboard control of selectors are broken in this version of Edge

uellue, 2020-06-29:
Not sure if we keep this in the 0.6 milestone, defer it to later or file it under "Known issue, please use a different browser"? As a side note, drag and drop of files into text fields on GitHub is also broken in that version...

sk1p, 2020-06-29:
I'm tempted to push this to later, as it is easily worked around. Global and Germany-wide usage of Edge is quite low (under 3%, version 16 under 0.03%), so unless we hear from other users, we should not invest a lot of time into support. We could of course create a policy of supported browsers and put in a list of browsers we test on into the docs, which currently boils down to Chromium-based (Chrome, new Edge, Opera, ... should all be similar enough) and Firefox.

uellue, 2020-06-29:
> create a policy of supported browsers and put in a list of browsers we test on into the docs

👍 That's definitely a good information to have!

uellue, 2023-08-15:
Seems to be fixed, GUI works fine with current Edge.

closed 2023-08-15 by uellue