LiberTEM/LiberTEM #836: Strange streaks in FFT analysis

uellue, 2020-06-29:
How to reproduce:

* LiberTEM version 0.5.0
* Use FFT analysis in the GUI on D:\Merlin_data\Janghyun\20200619 - SNBD Test\01_256X256_1ms_TV120/Test.hdr

Result:

[image]

The streaks look like artifacts.

sk1p, 2020-06-29:
Did you confirm that there is no artifact in the source data? Can you make this data set available to me, or should I have a look when I'm at the institute again?

uellue, 2020-06-29:
> Did you confirm that there is no artifact in the source data?

I didn't go in depth with scripting. The leftmost panel looked very different between the positions with the streaks and the other areas, while the central "pick" frame looked the same. Janghyun told me that there was building work at the time, although I don't know how that should have caused the anomaly.

> Can you make this data set available to me, or should I have a look when I'm at the institute again?

Currently it is only on the Merlin PC. Maybe you or I can have a look when we are at the institute.

sk1p, 2020-06-29:
> The leftmost panel looked very different between the positions with the streaks and the other areas

I think we need to have a detailed look at the absolute result values here to be sure

uellue, 2020-11-12:
I could reproduce "funny" streaks in our `20200518 165148` Ptychography dataset:

[image]

The analysis runs quite unstable in a Jupyter notebook: It randomly hangs and crashes. This looks like a deeper issue?

uellue, 2020-11-12:
Tracked it down: It is a bad pixel in the detector.

[image]

The instability seems to be caused by other factors, not sure what is going on.

uellue, 2020-11-12:
Closing here since the issue is not related to LiberTEM, but an artifact in the data.

closed 2020-11-12 by uellue