LiberTEM/LiberTEM #838: Opening an MIB file from HDD is slow with current master

uellue, 2020-06-29:
How to reproduce:

* LiberTEM current master
* Windows 10 (Merlin PC)
* Stored on a HDD that can reach about 100 MB/s write speed in a 1:1 copy, as an approximate performance measure.
* MIB file 256x256x256x256 with all frames in a single file
* D:\Merlin_data\Janghyun\20200619 - SNBD Test\04_256X256_1ms_TV160/Test.hdr
* File is not in the FS cache
* Open with LiberTEM GUI and add basic analysis

What happens:

* Read performance 7-8 MB/s, unusably slow

Expected: Reaches at least the read performance of LiberTEM 0.5.0, about 70 MB/s

If the file is in the FS cache, read performance is normal (good).

Possibly related to #814?

sk1p, 2020-06-29:
> If the file is in the FS cache, read performance is normal (good).

This points to the seek patterns for tiled reading being the problem - apparently Windows doesn't recognize those as "linear" and doesn't perform any read-ahead. I think we are at a point where we should make a general decision how to handle this - if we want to have usable performance on slow-seek media, we need to read in a completely different way than if we want to have optimal performance on fast SSDs. On an NVMe SSD, we theoretically can increase performance by dispatching even more parallel read commands (using liburing etc.), possibly reading deeper tiles, which would completely tank HDD perf.

I think it may very well be worthwhile to explore this, as other I/O "backends" can have similar issues, like reading from parallel file systems, NFS, or via HTTP. I'll see that I do some local testing soon, and then we should discuss how we prioritize a fix for this issue.

uellue, 2020-06-29:
> we should make a general decision how to handle this - if we want to have usable performance on slow-seek media

On the one hand, since performance was acceptable before and people including us still use such media, we should at least be on par with 0.5.0. However, we should also keep the massive performance gains of optimized tiled processing.

Having different code paths for slow-seek and fast-seek media could get messy, though. More code volume, and how to decide which one to use? Are there OS interfaces that can tell on what kind of storage a file is kept? Should we let users decide?

uellue, 2020-06-29:
By the way, can this be helpful for mmapped data, perhaps?

* https://linux.die.net/man/2/fadvise
* https://docs.microsoft.com/en-us/windows/win32/api/memoryapi/nf-memoryapi-prefetchvirtualmemory?redirectedfrom=MSDN

uellue, 2020-06-29:
In any case, a good use case for the benchmarking infrastructure... #819

sk1p, 2020-06-29:
> On the one hand, since performance was acceptable before and people including us still use such media, we should at least be on par with 0.5.0. However, we should also keep the massive performance gains of optimized tiled processing.

> how to decide which one to use

I think we only have a chance of making the user decide this (i.e. opt in to the "slow HDD" I/O backend) - I don't think there is a portable API to find expected performance for block devices. On Linux, this is definitely possible to find out, for example in `/sys/block/sda/queue/` on my system there is a file `rotational`, and there are also I/O statistics available, like average I/O latency ("total wait time for read requests" and "number of read I/Os processed" [from `/sys/block/*/stat`](https://www.kernel.org/doc/Documentation/block/stat.txt)). But automatically deciding could also make performance unpredictable...

Maybe we can get some inspiration from data bases here - they have tuning parameters that specify how costly some operations are, for example [`random_page_cost` on PostgreSQL](https://www.postgresql.org/docs/12/runtime-config-query.html) can be changed to tip the scales between doing a sequential scan over the whole table, vs. doing more "random" reads using indexes. With a HDD it can be faster to "waste" bandwidth and scan the whole table than having to seek repeatedly to find specific values, depending on what percentage of the whole table we expect to hit with the query. Of course, this also depends on the likelyhood of hitting the filesystem cache, ...

sk1p, 2020-06-29:
> https://linux.die.net/man/2/fadvise

Yeah, see here: https://github.com/LiberTEM/LiberTEM/blob/cf241433b094473c9edc38e80c8d79f4a9e8ffac/src/libertem/io/dataset/base/backend.py#L240

Could be made a bit smarter about the values of the `roi`, but in general it is there.

> https://docs.microsoft.com/en-us/windows/win32/api/memoryapi/nf-memoryapi-prefetchvirtualmemory?redirectedfrom=MSDN

Thanks for the link! I'll have a look if there is a good way to use this from Python...

sk1p, 2020-06-29:
See also discussion here: https://bugs.python.org/issue32941

uellue, 2020-06-29:
> Yeah, see here:

👍 nice, didn't know it was implemented already!

sk1p, 2020-06-29:
> In any case, a good use case for the benchmarking infrastructure... #819

How would that work, can we do a run of I/O benchmarks on different storage locations on the same host and have them distinguishable in the result history?

uellue, 2020-06-30:
> can we do a run of I/O benchmarks on different storage locations on the same host and have them distinguishable in the result history?

We can use `pytest.mark.parametrize` and/or [add specific metadata](https://pytest-benchmark.readthedocs.io/en/latest/usage.html#extra-info). One could, for example, add the full path or environment variables with a base directory to `extra_info`. That way one can distinguish separate runs on different locations.

uellue, 2020-10-08:
A few thoughts on how to handle this:

The optimal IO strategy depends on seek time, optimal block size, transfer rate and the ability to handle concurrent requests efficiently. In reality we may have the following sources when reading: File system cache, SSD, HDD, and network

* File system cache: No seek time, block size is single cache line, highest possible transfer rate, handles concurrent requests nearly instantaneous.
  - Optimal strategy: Memory mapping, direct access of exactly the data that is needed when it is needed.
* SSD: Very low seek time independent of jump size, optimal block size in the kB to MB range(?), transfer rate high, speed increase from concurrent requests.
  - Optimal strategy: Tiled reading (concurrent read of several blocks with 10s to 100s of kB each) should work well. Direct IO or Liburing for files larger than RAM. Memory mapping and reading exactly what is needed when it is needed should work reasonably well. fadvise() in combination with memory mapping should help a bit if files fit into RAM to avoid overheads from page faults.
* HDD: Large seek time, block size 4k (?), moderate transfer rate, sequential reads are highly favored, doesn't handle concurrency well at all.
  - Optimal strategy: Sequential read of contiguous portions. Bridging a few "useless" MB vs. discontinuous reads probably favored (seek time vs. transfer rate), further tiling / disassembly from memory buffer. Memory mapping *only* if fadvise() or the Windows equivalent work reliably to let the OS prefetch enough data into the cache with sequential reads so that our memory access patterns don't trigger page faults.
* Network: Moderate to large latency (software stack latency + network latency + latency of storage), small to medium transfer rate, may or may not handle concurrent access well, may have overheads when accessing small blocks -- we don't know what is on the other side!
  - Optimal strategy: Avoid page faults when memory mapping (round trip time), i.e. use only if fadvise() can avoid them reliably. Probably easiest to read larger contiguous blocks at a time similar to HDD. Saving bandwidth may, in tendency, be more favorable compared to HDD.

The way I can see it, the decision tree boils down to the following items:
* Is the file in the file system cache? If yes, mmap() wins by a margin.
* Is the file on an SSD? If yes, mmap() will do reasonably well under all circumstances.
* Otherwise, reading larger blocks into memory will do better. SSDs will work well,, too.

Reading large blocks will do reasonably well under all scenarios, while reading small portions with mmap() can have dramatic negative effects on HDD and possibly network, as illustrated by this Issue. That means the large block strategy can be a good failsafe.

Since it would be hard to find out what the current situation is beforehand and it changes dynamically (in the FS cache vs. outside, other processes doing things), we could implement a simple heuristic that switches between mmap() and reading large blocks: We simply monitor the effective read rate for each tile, and use whatever works better.

As an example, we could process three tiles with each mmap() and large contiguous read(). The first ones are warm-up, and we just use what gave the best average result for the remaining two. Furthermore, we can monitor the read rate. If it falls below, say, 50 MB/s per worker, we switch to the large block strategy. That one will always be a safe bet, and at 50 MB/s the additional memory transfer will have hardly any effect.

Possibly we can even keep it simpler: First tile mmap() warmup (Numba compilation etcetcetc). Then continue with mmap() and monitor. If below 50 MB/s, switch to large blocks for rest of partition.

Possibly we can also let the user override the read strategy as desired, in case the user is competent and the heuristic is not optimal. That's similar to the "direct IO" option we currently have. Options are mmap(), large blocks, DIO, and possibly liburing in the future (refs #268).

uellue, 2020-10-08:
For testing, we could have paths for the same data on different storage locations determined by, for example environment variables or a sort of "host file". That way we can run benchmarks with `pytest.mark.parametrize()` on all the storage locations in one run with the same tests and skip the ones where we don't have a file on that system.

Proposal for a reasonable order:

1. Make it reproducible and testable, for example using Moellenstedt
2. See if we can get the "Windows fadvise()" going and if it already resolves the issue.

If it doesn't:

3. Implement "large contiguous read" strategy at least for MIB, confirm effective remedy of this issue for HDD.
4. Implement user control over read strategy, which will facilitate further benchmarking
5. Implement "large contiguous read" strategy for other dataset types, if not already done in 2
6. Implement rate monitoring and dynamic switching of read strategy

Perhaps I can already start to get benchmarks going on different storage locations, at the same time switching over to our new data management directory structure?

sk1p, 2020-10-14:
As mentioned in  #887, this appears to be a Windows-specific problem.

Can you try to run the benchmarks with `fadvise` disabled? Just adding a `return` statement [here](https://github.com/LiberTEM/LiberTEM/blob/16011aee5fab65fc2a4094ddb4e1a203f5d51b3e/src/libertem/io/dataset/base/backend.py#L240) should do the trick.

We should see if not using `fadvise` on Linux means we get the same problem, and then decide if we want to implement a Windows-specific `fadvise` replacement, or if we want to implement buffered reading and enable it by default for Windows only.

uellue, 2020-10-14:
Linux generally doesn't seem to be affected even without fadvise(), at least with this file:

<details>

```
----------------------------------------------------------------------------------------------------------------------------- benchmark: 18 tests ------------------------------------------------------------------------------------------------------------------------------
Name (time in s)                                                                                                      Min                Max               Mean            StdDev             Median               IQR            Outliers     OPS            Rounds  Iterations
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_mask[/cachedata/users/libertem/reference/-cold_cache-cold_executor]                                           9.7464 (56.70)     9.7464 (56.70)     9.7464 (56.70)    0.0000 (1.0)       9.7464 (56.70)    0.0000 (1.0)           0;0  0.1026 (0.02)          1           1
test_mask[/cachedata/users/libertem/reference/-cold_cache-warm_executor]                                           0.5772 (3.36)      0.5772 (3.36)      0.5772 (3.36)     0.0000 (1.0)       0.5772 (3.36)     0.0000 (1.0)           0;0  1.7326 (0.30)          1           1
test_mask[/cachedata/users/libertem/reference/-warm_cache-cold_executor]                                           9.8039 (57.03)     9.8039 (57.03)     9.8039 (57.03)    0.0000 (1.0)       9.8039 (57.03)    0.0000 (1.0)           0;0  0.1020 (0.02)          1           1
test_mask[/cachedata/users/libertem/reference/-warm_cache-warm_executor]                                           0.4973 (2.89)      0.4973 (2.89)      0.4973 (2.89)     0.0000 (1.0)       0.4973 (2.89)     0.0000 (1.0)           0;0  2.0109 (0.35)          1           1
test_mask[/data/users/libertem/reference/-cold_cache-cold_executor]                                               12.7268 (74.04)    12.7268 (74.04)    12.7268 (74.04)    0.0000 (1.0)      12.7268 (74.04)    0.0000 (1.0)           0;0  0.0786 (0.01)          1           1
test_mask[/data/users/libertem/reference/-cold_cache-warm_executor]                                                4.4003 (25.60)     4.4003 (25.60)     4.4003 (25.60)    0.0000 (1.0)       4.4003 (25.60)    0.0000 (1.0)           0;0  0.2273 (0.04)          1           1
test_mask[/data/users/libertem/reference/-warm_cache-cold_executor]                                                9.4579 (55.02)     9.4579 (55.02)     9.4579 (55.02)    0.0000 (1.0)       9.4579 (55.02)    0.0000 (1.0)           0;0  0.1057 (0.02)          1           1
test_mask[/data/users/libertem/reference/-warm_cache-warm_executor]                                                0.5186 (3.02)      0.5186 (3.02)      0.5186 (3.02)     0.0000 (1.0)       0.5186 (3.02)     0.0000 (1.0)           0;0  1.9282 (0.33)          1           1
test_mask[/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/-cold_cache-cold_executor]      9.8938 (57.56)     9.8938 (57.56)     9.8938 (57.56)    0.0000 (1.0)       9.8938 (57.56)    0.0000 (1.0)           0;0  0.1011 (0.02)          1           1
test_mask[/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/-cold_cache-warm_executor]      1.3670 (7.95)      1.3670 (7.95)      1.3670 (7.95)     0.0000 (1.0)       1.3670 (7.95)     0.0000 (1.0)           0;0  0.7315 (0.13)          1           1
test_mask[/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/-warm_cache-cold_executor]      9.7071 (56.47)     9.7071 (56.47)     9.7071 (56.47)    0.0000 (1.0)       9.7071 (56.47)    0.0000 (1.0)           0;0  0.1030 (0.02)          1           1
test_mask[/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/-warm_cache-warm_executor]      0.5586 (3.25)      0.5586 (3.25)      0.5586 (3.25)     0.0000 (1.0)       0.5586 (3.25)     0.0000 (1.0)           0;0  1.7901 (0.31)          1           1
test_sequential[/cachedata/users/libertem/reference/-cold_cache]                                                   0.3766 (2.19)      0.3766 (2.19)      0.3766 (2.19)     0.0000 (1.0)       0.3766 (2.19)     0.0000 (1.0)           0;0  2.6554 (0.46)          1           1
test_sequential[/cachedata/users/libertem/reference/-warm_cache]                                                   0.1942 (1.13)      0.1942 (1.13)      0.1942 (1.13)     0.0000 (1.0)       0.1942 (1.13)     0.0000 (1.0)           0;0  5.1489 (0.89)          1           1
test_sequential[/data/users/libertem/reference/-cold_cache]                                                        3.5982 (20.93)     3.5982 (20.93)     3.5982 (20.93)    0.0000 (1.0)       3.5982 (20.93)    0.0000 (1.0)           0;0  0.2779 (0.05)          1           1
test_sequential[/data/users/libertem/reference/-warm_cache]                                                        0.1719 (1.0)       0.1719 (1.0)       0.1719 (1.0)      0.0000 (1.0)       0.1719 (1.0)      0.0000 (1.0)           0;0  5.8175 (1.0)           1           1
test_sequential[/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/-cold_cache]              1.1287 (6.57)      1.1287 (6.57)      1.1287 (6.57)     0.0000 (1.0)       1.1287 (6.57)  [... cut ...]
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

</details>

Next step would be reproducing on Windows, and if this doesn't reproduce the issue then test with the exact same file that created the issue. The test file is smaller and split up in small MIB files, while the issue was caused by a larger monolithic MIB file.

uellue, 2020-10-14:
In any case, the results with cold cache and fadvise() were all a bit better than the ones without fadvise(), so it DOES help some.

sk1p, 2020-10-14:
Thanks for testing this! It would be good to track these numbers over time. 

> Next step would be reproducing on Windows, and if this doesn't reproduce the issue then test with the exact same file that created the issue. The test file is smaller and split up in small MIB files, while the issue was caused by a larger monolithic MIB file.

Let me know if you need any help with that. We can also create synthetic test data if needed (it's easy to create a large MIB file from many small ones, for example).

uellue, 2020-10-14:
Benchmark results with Windows:

<details>

```
--------------------------------------------------------------------------------------------------------- benchmark: 6 tests --------------------------------------------------------------------------------------------------------
Name (time in s)                                                           Min                Max               Mean            StdDev             Median               IQR            Outliers     OPS            Rounds  Iterations
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_mask[C:/Users/dwebe/Desktop/reference-cold_cache]                 62.7528 (91.98)    62.7528 (91.98)    62.7528 (91.98)    0.0000 (1.0)      62.7528 (91.98)    0.0000 (1.0)           0;0  0.0159 (0.01)          1           1
test_mask[C:/Users/dwebe/Desktop/reference-warm_cache]                  0.8421 (1.23)      0.8421 (1.23)      0.8421 (1.23)     0.0000 (1.0)       0.8421 (1.23)     0.0000 (1.0)           0;0  1.1876 (0.81)          1           1
test_mask_firstrun[C:/Users/dwebe/Desktop/reference-cold_executor]     13.0089 (19.07)    13.0089 (19.07)    13.0089 (19.07)    0.0000 (1.0)      13.0089 (19.07)    0.0000 (1.0)           0;0  0.0769 (0.05)          1           1
test_mask_firstrun[C:/Users/dwebe/Desktop/reference-warm_executor]      0.8810 (1.29)      0.8810 (1.29)      0.8810 (1.29)     0.0000 (1.0)       0.8810 (1.29)     0.0000 (1.0)           0;0  1.1351 (0.77)          1           1
test_sequential[C:/Users/dwebe/Desktop/reference-cold_cache]            8.7860 (12.88)     8.7860 (12.88)     8.7860 (12.88)    0.0000 (1.0)       8.7860 (12.88)    0.0000 (1.0)           0;0  0.1138 (0.08)          1           1
test_sequential[C:/Users/dwebe/Desktop/reference-warm_cache]            0.6823 (1.0)       0.6823 (1.0)       0.6823 (1.0)      0.0000 (1.0)       0.6823 (1.0)      0.0000 (1.0)           0;0  1.4657 (1.0)           1           1
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

</details>

The issue is clearly reproduced with this dataset and benchmark on Windows. The difference between Linux and Windows is striking.

uellue, 2020-10-14:
> Thanks for testing this! It would be good to track these numbers over time.

Yes, pytest-benchmarks can do that and I believe we already have a few "dummy" runs. We postponed regular tests until we have more benchmarks. Now might be a good time to get started with that. :-) Perhaps we can discuss a strategy how to organize results and make them comparable?

uellue, 2020-10-15:
For posterity, this code _should_ prefetch memeory, but doesn't improve the benchmark results or triggers disk activity if the cache is empty. Not sure if I called it right, though.

```python
class MEMORY_RANGE_ENTRY(ctypes.Structure):
    _fields_ = [
        ("VirtualAddress", ctypes.c_void_p),
        ("NumberOfBytes", ctypes.c_size_t),
    ]

def prefetch_mmap(memmap):
    first = ctypes.c_char_p(memmap[:1])
    mentry = MEMORY_RANGE_ENTRY(
        VirtualAddress=ctypes.cast(first, ctypes.c_void_p),
        NumberOfBytes=memmap.size(),
    )
    result = ctypes.windll.kernel32.PrefetchVirtualMemory(
        ctypes.windll.kernel32.GetCurrentProcess(),
        ctypes.c_ulong(1),
        mentry,  # or perhaps ctypes.c_void_p(ctypes.addressof(mentry))
        ctypes.c_ulong(0)
    )
    if result == 0:
        raise RuntimeError("Return code 0 means error, see https://docs.microsoft.com/en-us/windows/win32/api/memoryapi/nf-memoryapi-prefetchvirtualmemory")

```

uellue, 2020-10-15:
Since the issue is reproduced on Windows and prefetching didn't work, we'll have to implement the "large block" reading strategy, it seems.

Linux seems to do an excellent job to translate the parallel tiled memory access pattern into larger contiguous disk reads, while Windows produces a flurry of seeks.

uellue, 2020-10-15:
As an alternative to changing the reading strategy in this release, perhaps one could change the tiling scheme instead to use the same scheme as 0.5 if IO performance is slow? That could be done per partition. Possibly the UDFRunner could raise an "IOTooSlowExeption" when it detects slow IO after processing a few tiles,  and start over with tiling that will give sequential access in run_for_partition()?

The real sequential reading strategy could then be implemented in #473.

closed 2021-01-25 by uellue

uellue, 2021-02-05:
As a follow-up comment, the current buffered reader in 0.6.0rc2 reaches about 30-40 % of the theoretical read performance of the HDD in the "mobile workstation" machine with the 4 GB EMPAD dataset. IMO that's good enough for now, one could just see if #473 can perhaps beat that at some point. :-)

The performance impact of buffered reading vs memmap for files in the cache is quite noticeable on that machine, by the way. The CPU in that machine [has very poor memory bandwidth](https://ark.intel.com/content/www/us/en/ark/products/89608/intel-xeon-processor-e3-1505m-v5-8m-cache-2-80-ghz.html#tab-blade-1-0-3). It seems that the additional copy operation has a big impact on performance for that reason.