LiberTEM/LiberTEM #845: Copy remote cluster connection code

twentyse7en, 2020-07-13:
If GUI is successfully connected to a remote cluster, the connection code will be
generated and an option to copy the code. Where would the ideal place for providing the option?

sk1p, 2020-07-20:
I think this could be combined with a status display for the current cluster connection. Maybe in the menu bar somewhere? One option would be to put the "action"-like items ("About", "Shutdown") into a drop-down menu and display the cluster status in the top-right, where the shutdown button is right now. What do you think?

twentyse7en, 2020-07-20:
Instead, can we put the cluster status to the left of "Shutdown"? What are the information to be displayed in the status?

sk1p, 2020-07-20:
> Instead, can we put the cluster status to the left of "Shutdown"?

Could be possible - we just need to make sure the menu doesn't get too crowded, or, in other words, that the menu is stays clear and usable.

> What are the information to be displayed in the status?

Hmm, something like "connected to (local cluster|cluster URI), N CPU workers, M GPUs"? If you can think of a way to show this more concise, that would also be great. Then maybe a "copy" icon to show your copying component?

PR #863 (2020-11-06): WIP: Display cluster conn details
If GUI is successfully connected to a remote cluster, the connection code will be
generated and an option to copy the code.  Fixes #845 

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [ ] I have added/updated test cases
* [x] I have included  [... cut]
closed 2020-11-06 by sk1p