LiberTEM/LiberTEM #846: Make sure size calculations don't overflow

uellue, 2020-07-13:
`np.prod()` is currently used all over the LiberTEM code base to calculate the number of entries based on the shape of array-like objects, like datasets, partitions, tiles, slices etc. [`np.prod()` upcasts integers to `np.int`](https://numpy.org/doc/stable/reference/generated/numpy.prod.html), [which is 32 bit on Windows even on 64 bit](https://github.com/numpy/numpy/issues/9464). This is often too small to hold the number of pixels in datasets.  This already caused two overflow bugs (#493 #495 #844) in LiberTEM.

We should make sure that this can't happen so easily for a third time. Aspects could include:

* Add a method or property to relevant objects to calculate the size safely, similar to https://numpy.org/doc/stable/reference/generated/numpy.ndarray.size.html that uses Python arbitrary size integers.
* Make sure we don't use `np.prod()` unsafely in the code base.
* Introduce a dedicated `size_prod()` function that upcasts to int64.
* Monkey patch `np.prod()` to always upcast to `np.int64` instead of `np.int`.
  * How to deal with Numba? Overloading `np.prod()`?

uellue, 2020-07-20:
Bumped to 0.7 since @AnandBaburajan fixed the issue on the short end in #844 and we want to keep the scope of 0.6 manageable.

sayandip18, 2020-10-28:
>   * Introduce a dedicated `size_prod()` function that upcasts to int64.
>   * Monkey patch `np.prod()` to always upcast to `np.int64` instead of `np.int`.

```
def size_prod(a, axis=None, dtype=None, out=None, keepdims=np._NoValue, initial=np._NoValue, where=np._NoValue):
	return np.prod(a, axis=None, dtype=np.int64, out=None, keepdims=np._NoValue, initial=np._NoValue, where=np._NoValue)
```

How about defining something like this which takes the [exact same arguments](https://github.com/numpy/numpy/blob/v1.19.0/numpy/core/fromnumeric.py#L2881-L3000) as `np.prod()`? We need to replace all instances of `np.prod()` with this function in that case.

uellue, 2021-11-05:
In 3.8 [`math.prod`](https://docs.python.org/3/library/math.html#math.prod) was added, which uses infinite width integers. This would be the perfect choice for size calculations.

Possibly we can add our own `prod` implementation for versions below 3.8?

PR #1145 (2021-11-11): Replace NumPy `prod()` with infinite width integer `prod()`
Avoids integer overflows in size calculations

Closes #846, see there for more detail

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [N/A] I have included the [rebuilt producti [... cut]
commit 79fcd37b (2021-11-11): Replace NumPy `prod()` with infinite width integer `prod()`

Avoids integer overflows in size calculations

Closes #846, see there for more detail
closed 2021-11-11 by sk1p