LiberTEM/LiberTEM #849: Ad-hoc DataSet for custom/weird formats

sk1p, 2020-07-20:
A stable interface for reading custom file formats. Motivation: load things like output from simulations that can be in weird formats, like a directory full of matlab files, or a list of tiff images, a list of raw files etc. - mostly for "list of files"-type datasets, like #206. This is only meant to make it possible to load these in LiberTEM in an ad-hoc way, not really with performance in mind.

As @uellue said: "a UDF for DataSets", but maybe going more into the direction of `Context.map` instead of the full UDF interface.

Discussion: to get reasonable performance while still being easy to use, I would say we should concentrate on simple full-frame I/O. How flexible should the interface be? Should we only support file formats with one frame per file, or should we support more arbitrary mappings?

Possible options:

1) Input: ordered list of files, shaping parameters, function: filename → `sig`-shaped numpy array
2) Input: shaping parameters, `**kwargs`, function: index → `sig`-shaped numpy array

For option 2), basically arbitrary data sets would be supported, but the function needs to handle some index translation, option 1) would only support the "directory of files" use-case, but would be a bit easier to use.

```python
def option_1(fn):
    mat_data = scipy.io.loadmat(fn)
    return mat_data['some_interesting_array']

ds = AdHocFunctionDataSet(files=[...], sig_shape=..., nav_shape=..., load=option_1)

# files and images_per_file are passed as **kwarg from the DataSet to the load function
def option_2(idx, files, images_per_file):
    fn = files[idx // images_per_file]  # or some other mapping
    mat_data = scipy.io.loadmat(fn)
    return mat_data['some_interesting_array'][idx % images_per_file]

ds = AdHocFunctionDataSet(files=[...], images_per_file=17, sig_shape=..., nav_shape=..., load=option_2)
```

In option 2, we could also support formats where there is not only a single image per file, but some kind of chunking was applied when saving, as shown in the example above (I think some multi slice simulation software works in this way...). The index would be into the (flat?) navigation axis.

Related to #201, #206

sk1p, 2020-07-20:
This is mostly for keeping the idea/interface organized, it can be implemented when it becomes inconvenient to use alternatives: until now, I converted folder-of-files data sets into a single raw file, but this may be inconvenient if there are many of these data sets, or if they are very large so storage becomes an issue...

sk1p, 2020-07-20:
Oh, this could also be (ab-)used for generating virtual data sets, like on-the-fly simulation, at least if we implement option 2

uellue, 2020-07-20:
> Should we only support file formats with one frame per file, or should we support more arbitrary mappings?

The last format I saw was one Matlab file per simulated row or column, so I'd say more arbitrary mappings should be possible.

Single frame processing should be ideal since that keeps it simple and it should be compatible with any processing method in the UDF interface. We should just make sure that `process_partition()` actually receives a partition by accumulating frames in a buffer since otherwise ROI partitions would not work with such datasets.

uellue, 2020-07-20:
> by accumulating frames in a buffer

Hm, one could support proper tiling as well that way. As long as the reading function is fast, this should not be too bad, actually.

> Context.map

Creating task data and such could be useful, for example to open files that hold many frames only once?

uellue, 2021-03-18:
#922 would be a good way to do this: Just create a Dask array with `dask.delayed` and use that as a dataset.

uellue, 2021-12-22:
> #922 would be a good way to do this: Just create a Dask array with `dask.delayed` and use that as a dataset.

Done in #1137, closing

closed 2021-12-22 by uellue