LiberTEM/LiberTEM #854: ROI not used in generated template for SD analysis

uellue, 2020-07-24:
In notebook with SD analysis and circular ROI:

```
nx, ny = ds.shape.nav
roi = masks.circular(centerX=22.367344631814362, centerY=75.2197918337658, imageSizeX=nx, imageSizeY=ny, radius=11.25)
sd_udf  = StdDevUDF()
sd_result = ctx.run_udf(dataset=ds, udf=sd_udf)
```

The ROI is generated correctly, but not used when running the UDF.

uellue, 2020-07-24:
Probably one should check other analyses that use the ROI as well, namely sum.

twentyse7en, 2020-07-25:
why [`random_hdf5`](https://github.com/LiberTEM/LiberTEM/blob/master/conftest.py#L66-L69) has `np.ones((5, 5, 16, 16))` as data ?

sk1p, 2020-07-27:
> why [`random_hdf5`](https://github.com/LiberTEM/LiberTEM/blob/master/conftest.py#L66-L69) has `np.ones((5, 5, 16, 16))` as data ?

My guess: it's a copy/paste error. Good catch! If you change it to generate random data instead, do any of the tests break?

twentyse7en, 2020-07-27:
changed to `np.random.randint((5, 5, 16, 16))` , four tests failed in `test_hdf5.py`
<details>
<summary> Error report </summary>

```
GLOB sdist-make: /home/abi/LiberTEM/setup.py
py37 recreate: /home/abi/LiberTEM/.tox/py37
py37 installdeps: -rtest_requirements.txt, -roverride_requirements.txt
py37 inst: /home/abi/LiberTEM/.tox/.tmp/package/8/libertem-0.6.0.dev0.zip
py37 installed: aiohttp==3.6.2,apipkg==1.5,async-timeout==3.0.1,attrs==19.3.0,backcall==0.2.0,bleach==3.1.5,chardet==3.0.4,click==7.1.2,cloudpickle==1.5.0,coverage==5.2.1,cycler==0.10.0,Cython==0.29.21,dask==2.21.0,decorator==4.4.2,defusedxml==0.6.0,distributed==2.21.0,entrypoints==0.3,execnet==1.7.1,h5py==2.10.0,hdbscan==0.8.26,HeapDict==1.0.1,idna==2.10,imageio==2.9.0,importlib-metadata==1.7.0,ipykernel==5.3.4,ipython==7.16.1,ipython-genutils==0.2.0,jedi==0.17.2,Jinja2==2.11.2,joblib==0.16.0,jsonschema==3.2.0,jupyter-client==6.1.6,jupyter-core==4.6.3,kiwisolver==1.2.0,libertem @ file:///home/abi/LiberTEM/.tox/.tmp/package/8/libertem-0.6.0.dev0.zip,libertem-blobfinder==0.4.1,llvmlite==0.33.0,MarkupSafe==1.1.1,matplotlib==3.3.0,mistune==0.8.4,more-itertools==8.4.0,msgpack==1.0.0,multidict==4.7.6,nbconvert==5.6.1,nbformat==5.0.7,ncempy==1.6.0,networkx==2.4,numba==0.50.1,numpy==1.19.1,packaging==20.4,pandocfilters==1.4.2,parso==0.7.1,pexpect==4.8.0,pickleshare==0.7.5,Pillow==7.2.0,pluggy==0.13.1,primesieve==2.2.0,prompt-toolkit==3.0.5,psutil==5.7.2,ptyprocess==0.6.0,py==1.9.0,py-cpuinfo==7.0.0,Pygments==2.6.1,pyparsing==2.4.7,pyrsistent==0.16.0,pytest==5.4.3,pytest-asyncio==0.14.0,pytest-benchmark==3.2.3,pytest-cov==2.10.0,pytest-forked==1.2.0,pytest-xdist==1.33.0,python-dateutil==2.8.1,PyWavelets==1.1.1,PyYAML==5.3.1,pyzmq==19.0.1,scikit-image==0.17.2,scikit-learn==0.23.1,scipy==1.5.2,six==1.15.0,sortedcontainers==2.2.2,sparse==0.10.0,tblib==1.7.0,testpath==0.4.4,threadpoolctl==2.1.0,tifffile==2020.7.24,toolz==0.10.0,tornado==6.0.4,tqdm==4.48.0,traitlets==4.3.3,typing-extensions==3.7.4.2,wcwidth==0.2.5,webencodings==0.5.1,websockets==8.1,yarl==1.5.0,zict==2.0.0,zipp==3.1.0
py37 run-test-pre: PYTHONHASHSEED='1859303691'
py37 run-test: commands[0] | pytest --durations=10 --cov=libertem --cov-report=term --cov-report=html --cov-report=xml tests/
====================================== test session starts =======================================
platform linux -- Python 3.7.4, pytest-5.4.3, py-1.9.0, pluggy-0.13.1
cachedir: .tox/py37/.pytest_cache
benchmark: 3.2.3 (defaults: timer=time.perf_counter disable_gc=False min_rounds=5 min_time=0.000005 max_time=1.0 calibration_precision=10 warmup=True warmup_iterations=100000)
rootdir: /home/abi/LiberTEM, inifile: pytest.ini
plugins: cov-2.10.0, asyncio-0.14.0, xdist-1.33.0, benchmark-3.2.3, forked-1.2.0
collected 758 items / 20 deselected / 738 selected

tests/test_analysis_base.py ..                                                             [  0%]
tests/test_analysis_clust.py .                                                             [  0%]
tests/test_analysis_com.py ..............................                                  [  4%]
tests/test_analysis_masks.py x............................................................ [ 12%]
..............                                                                             [ 14%]
tests/test_analysis_radialfourier.py ..............                                        [ 16%]
tests/test_analysis_raw.py ........................                                        [ 19%]
tests/test_analysis_rawfft.py ..                                                           [ 20%]
tests/test_analysis_sd.py .                                                                [ 20%]
tests/test_analysis_shapes.py ......................                                       [ 23%]
tests/test_analysis_sum.py ............                                                    [ 24%]
tests/test_analysis_sumfft.py ..                                                           [ 25%]
tests/test_analysis_sumsig.py .                                                            [ 25%]
tests/test_context.py .                                                                    [ 25%]
tests/test_fileowner.py .                                                                  [ 25%]
tests/test_fs.py s..                                                                       [ 25%]
tests/test_fullmatch.py .....                                                              [ 26%]
tests/test_gridmatching.py ........                                                        [ 27%]
tests/test_job_deprecation.py ..                                                           [ 27%]
tests/test_local_cluster.py ..ss.                                                          [ 28%]
tests/test_masks.py ......                                                                 [ 29%]
tests/test_smoke.py .                                                                      [ 29%]
tests/test_utils.py ........                                                               [ 30%]
tests/test_viz.py ............                                                             [ 32%]
tests/common/test_bufferwrapper.py ....                                                    [ 32%]
tests/common/test_mask_container.py .......                                                [ 33%]
tests/common/test_shape.py ...............                                                 [ 35%]
[... cut ...]
FAILED tests/io/datasets/test_hdf5.py::test_roi_2[4] - ValueError: not enough values to unpack ...
= 4 failed, 614 passed, 117 skipped, 20 deselected, 3 xfailed, 133 warnings in 1068.98s (0:17:48) =
ERROR: InvocationError for command /home/abi/LiberTEM/.tox/py37/bin/pytest --durations=10 --cov=libertem --cov-report=term --cov-report=html --cov-report=xml tests/ (exited with code 1)
____________________________________________ summary _____________________________________________
ERROR:   py37: commands failed
```

</details>

twentyse7en, 2020-07-27:
No, tests are working fine.