LiberTEM/LiberTEM #856: Test coverage for notebook generator

sk1p, 2020-07-27:
A reminder that the notebook generation of the following analyses is not yet (fully) covered by test cases:

- Clustering analysis (I think the code generation was not complete here, too?)
- Radial Fourier analysis (I think this is waiting for the empyre stuff?)
- Raw FFT (`PickFFTFrameAanlysis`)
- `SumSigAnalysis`

Test coverage can be checked via the `htmlcov` directory after running the tests via tox, or (soon!) via [the new azure pipelines pages](https://dev.azure.com/liberTEM/LiberTEM/_build/results?buildId=20&view=codecoverage-tab)

twentyse7en, 2020-07-28:
`Radial Fourier Analysis` notebook requires the `empyre`, so the test is currently skipped. `Cluster Analysis` notebook generation is incomplete.

sk1p, 2021-01-27:
I think this is done, closing.

closed 2021-01-27 by sk1p