LiberTEM/LiberTEM #859: Improve notebook generator

twentyse7en, 2020-07-28:
* [ ] Fix cluster analysis notebook generation
* [ ] Option to copy and paste cells instead of downloading

twentyse7en, 2020-07-30:
Is this general design okay?

[image]

sk1p, 2020-07-30:
Looks good to me!

As an addition, it could be nice to be able to copy the whole source code in one go, what do you think? Also, maybe provide a "download as Jupyter Notebook" button/link in the header of this modal, too?

twentyse7en, 2020-07-30:
> As an addition, it could be nice to be able to copy the whole source code in one go

If that's useful, no problem in adding that :+1:

sk1p, 2020-09-08:
@twentyse7en I think with #864 merged, this issue is done, right?

twentyse7en, 2020-09-09:
> @twentyse7en I think with #864 merged, this issue is done, right?

Yes

closed 2020-09-09 by sk1p