LiberTEM/LiberTEM #892: Update install documentation to use `python3 -m pip`

uellue, 2020-10-26:
See https://snarky.ca/why-you-should-use-python-m-pip/

For some reason, I have to do it like that on moellenstedt after Python was updated. Using `pip install <...>` tries to install with Python 2 in the wrong environment. That means this issue is real and updating our documentation can help to steer newer users away from potential trouble.

uellue, 2020-10-26:
Assigning to 0.6 since this is a relatively minor update and we can do it as part of documentation QA during the release process.

pragat-pandya, 2020-10-27:
I would like to work on this issue, Please assign this task to me!

sk1p, 2020-10-28:
> I would like to work on this issue, Please assign this task to me!

Sure thing, please let me know if you need any help!

PR #975 (2021-02-16): Use `python -m pip`
Closes #892

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [N/A] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases

## Reviewer Checklist:

* [N/A] `/azp run libertem.libertem-data` passed
commit d1f1a145 (2021-02-16): Use `python -m pip`

Closes #892
closed 2021-02-16 by sk1p