LiberTEM/LiberTEM #903: SumSigUDF on a stack of DM files fails

AnandBaburajan, 2020-11-23:
How to reproduce on the master branch:

```python
ds = ctx.load("dm", files=["/media/anand/X/0/DM4/a.dm4","/media/anand/X/0/DM4/b.dm4"])
ctx.run_udf(dataset=ds, udf=SumSigUDF())
```

`backend._get_tiles_straight` is used in this case where it looks like there's a bug while reading from multiple files.

I'm trying to fix this at #793

sk1p, 2020-11-23:
Thanks for reporting! Indeed, that looks like a bug: the situation should be caught by the check here and enable the alternative reading mode with a copy:

https://github.com/LiberTEM/LiberTEM/blob/e9f32332d5314dfd4c6c71c967532ca094fff18f/src/libertem/io/dataset/base/backend.py#L78-L83

Maybe the check is not completely right for the situation?

AnandBaburajan, 2020-11-23:
> enable the alternative reading mode with a copy

Oh, I thought I could fix it by looking into `_get_tiles_straight`. Thanks, I'll try to understand why we need a copy!

sk1p, 2020-11-23:
To make sure I understand your reproducer, how large are the individual files you are stacking (that is, what shape do they have)?

> I'll try to understand why we need a copy and look into the check again.

The idea is that you can't read, let's say, a `(16, 64, 1024)` tile into one contiguous chunk of data (`DataTile`) in one go, if you only have one frame (let's say `(1, 1024, 1024)`) in each file - you need to read the individual `(1, 64, 1024)` parts and copy them into a temporary buffer. Generally, the number of frames per file should be at least the `depth` of the tiling scheme.

In your case, it actually looks like the tiles generated are `(1,) + ds.shape.sig` (because `SumSigUDF` is a `process_frame` UDF). So in that case, reading without copying should actually be possible, sorry if this caused any confusion!

AnandBaburajan, 2020-11-24:
> How large are the individual files you are stacking (that is, what shape do they have)?

(1, 3838, 3710)

> Generally, the number of frames per file should be at least the `depth` of the tiling scheme. In your case, it actually looks like the tiles generated are `(1,) + ds.shape.sig`

Yep, thanks! I'll look into `_get_tiles_straight`.

sk1p, 2020-11-26:
After fixing #904 I also ran into this issue with a stack of multiple 3D DM files. You were right, the issue is in the `_get_tiles_straight` method. I quickly looked into it, and it seems that the method assumed single-file data sets, and didn't correct the slice into the memory map for the starting index of the current file.

PR #905 (2020-11-26): Fix problems with zero-copy reading and DM
Zero-copy reading was not well tested with multiple files, resulting in #904. Also, the DM reader didn't properly calculate the `start_idx` and `end_idx` with multiple files with more than one frame per file (#903).

Fixes #903, fixes #904

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [N/A] I have ad [... cut]
commit 699517e8 (2020-11-26): Take correct offset into account when reading without copy

Fixes #903
closed 2020-11-26 by sk1p