LiberTEM/LiberTEM #904: Analyses on a stack of more-than-one-frame-DM files fail

AnandBaburajan, 2020-11-25:
```
AttributeError: 'NoneType' object has no attribute 'search_start'
```

How to reproduce:

```python
ds = ctx.load("dm", scan_size=(5, 8), files=["/media/anand/X/0/dm/1_20.dm3", "/media/anand/X/0/dm/2_20.dm3"])
# 1_20.dm3 and 2_20.dm3 are files with shape (20, 3838, 3710)
```
Then run any analyses on `ds`. From the traceback, the bug seems to be in the `fileset` part.

<details>
<summary>Traceback</summary>

```python
~/z/LiberTEM/src/libertem/io/dataset/dm.py in get_partitions(self)
    243                 fileset=self._fileset,
    244                 start_frame=start,
--> 245                 num_frames=stop - start,
    246             )
    247 

~/z/LiberTEM/src/libertem/io/dataset/base/partition.py in __init__(self, meta, partition_slice, fileset, start_frame, num_frames)
    144         """
    145         super().__init__(meta=meta, partition_slice=partition_slice)
--> 146         self._fileset = fileset.get_for_range(start_frame, start_frame + num_frames - 1)
    147         self._start_frame = start_frame
    148         self._num_frames = num_frames

~/z/LiberTEM/src/libertem/io/dataset/base/fileset.py in get_for_range(self, start, stop)
     34         return new FileSet filtered for files having frames in the [start, stop) range
     35         """
---> 36         files = self._get_files_for_range(start, stop)
     37         return self._clone(
     38             files=files,

~/z/LiberTEM/src/libertem/io/dataset/base/fileset.py in _get_files_for_range(self, start, stop)
     49         """
     50         files = []
---> 51         for f in self.files_from(start):
     52             if f.start_idx > stop:
     53                 break

~/z/LiberTEM/src/libertem/io/dataset/base/fileset.py in files_from(self, start)
     57 
     58     def files_from(self, start):
---> 59         lower_bound, f = self._tree.search_start(start)
     60         for idx in range(lower_bound, len(self._files)):
     61             yield self._files[idx]

~/z/LiberTEM/src/libertem/io/dataset/base/utils.py in search_start(self, value)
     72             return self.left.search_start(value)
     73         else:
---> 74             return self.right.search_start(value)
     75 
     76     def __str__(self):

AttributeError: 'NoneType' object has no attribute 'search_start'
```
</details>

sk1p, 2020-11-25:
Thanks for the report. Is this with the current `master` branch? I'll try to reproduce this locally and get back to you.

AnandBaburajan, 2020-11-25:
> Is this with the current `master` branch?

Yup

sk1p, 2020-11-26:
I can reproduce the issue - it seems the implementation from #877 was not complete, and I think I know how to fix this.

PR #905 (2020-11-26): Fix problems with zero-copy reading and DM
Zero-copy reading was not well tested with multiple files, resulting in #904. Also, the DM reader didn't properly calculate the `start_idx` and `end_idx` with multiple files with more than one frame per file (#903).

Fixes #903, fixes #904

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [N/A] I have ad [... cut]
closed 2020-11-26 by sk1p