LiberTEM/LiberTEM #911: Allow specifying nav_shape, sig_shape and sync_offset for K2IS

AnandBaburajan, 2020-11-28:
Continuation of #793.

`nav_shape` and `sig_shape` can be easily implemented. `sync_offset` implementation was almost finished, check [this](https://github.com/LiberTEM/LiberTEM/pull/793#issuecomment-687870313) comment and [this](https://github.com/LiberTEM/LiberTEM/pull/793#discussion_r443501353) one.

sk1p, 2021-03-24:
Replying to [this comment](https://github.com/LiberTEM/LiberTEM/pull/793#issuecomment-687870313):

> Now I've a better understanding on how K2IS's sectors are synced. My implementation works for minor changes to the offset but has some issues for example when sync_offset is set to 0 and all the data without the shutter_active flag is read too, but I'm working on it.

I think having `sync_offset=0` return all frames from the beginning is actually the right behavior. Would this problem be possibly fixed by considering a different default value for `sync_offset`? If the default were `None`, it could signal that the shutter active flag should be used and `sync_offset` should be set automatically. Thoughts?

AnandBaburajan, 2021-03-25:
> I think having `sync_offset=0` return all frames from the beginning is actually the right behavior.

I agree.

> If the default were `None`, it could signal that the shutter active flag should be used and `sync_offset` should be set automatically. Thoughts?

Yup, that was the plan.

 > but has some issues for example when sync_offset is set to 0 and all the data without the shutter_active flag is read too

I should have better explained the issues - actually I was getting weird frames when I set an offset <= 0. But a negative offset should mean blank frames so something was wrong with my implementation. I'll definitely look into this while working on our K2 simulator!

sk1p, 2021-03-25:
> I'll definitely look into this while working on our K2 simulator!

Great! :+1:

PR #1019 (2021-05-27): K2IS: reshaping, sync offset and time series support
Adds support for K2IS time series data sets.
@sk1p / @uellue, if the implementation is correct, I can continue with fixing #911 in this PR itself.

Fixes #911 

## Contributor Checklist:

* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [x] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/contributing.html? [... cut]
closed 2021-05-27 by uellue