LiberTEM/LiberTEM #914: Python 3.9 support

sk1p, 2020-11-30:
Currently blocked by numba support (Python 3.9 changed some bytecode ops), which should be available soon™.

sk1p, 2020-12-12:
We should also think about dropping Python 3.6 support soon after, as numpy 1.20 will only support Python 3.7+

uellue, 2020-12-14:
> We should also think about dropping Python 3.6 support soon after, as numpy 1.20 will only support Python 3.7+

Do we depend on any of the new features of NumPy 1.20? We pitch LiberTEM as suitable for integration in 3rd party software. If you think of, for example, GMS with its ultra long life and infrequent updates on instrument computers, one should support older Python versions for as long as possible. Upgrading packages in an environment is relatively easy, but upgrading Python can be more complicated.

I'd propose keeping support for 3.6 until dropping it creates benefits for users. Examples would be if beneficial Numba features like improved JIT caching #798 are only supported on newer Python versions, or if we make good use of shared memory support in Python 3.8 in the future #335. Even then, if LiberTEM can still run on an older Python somehow, even with some shortcomings or missing features, being able to run it on an instrument PC with very infrequent updates would still be desirable.

On the other hand I understand that we can't run a full test suite on all operating systems and countless Python versions for each push. What about having two or three versions with "full support" runs, and marking selected tests for reduced "legacy support" that are run on older Python versions Linux-only? We could pick a few tests such as the server or notebook template tests that run through a larger part of the code base under near production conditions to make sure that core features still work.

At some point, when LiberTEM has become mature like NumPy and using older releases is no problem, it would be easier to drop older Python versions completely and suggest that users just stick to an older release. At the moment that would be a shame since 0.6 will be a very significant improvement.

sk1p, 2020-12-14:
> Do we depend on any of the new features of NumPy 1.20? We pitch LiberTEM as suitable for integration in 3rd party software. If you think of, for example, GMS with its ultra long life and infrequent [...]

Okay, you have convinced me that it is worth it to support Python 3.6 as long as we can; especially given the outdated environments. So the policy would be to throw "not supported" errors at runtime, only when trying to access a specific feature, and to always allow the basic functionality to continue to work with older versions. After end-of-life of a legacy Python version, we would use "progressive enhancement" to allow new features to run on new Python versions, without guarantees for these features to run on old versions. A caveat for this approach is that we will be stuck with Python 3.6 compatible syntax for as long as we still keep it as legacy-supported. From the current perspective (taking 3.10 into account), I'm fine with this.

> On the other hand I understand that we can't run a full test suite on all operating systems and countless Python versions for each push. What about having two or three versions with "full support" [...]

Reducing the size of the CI matrix was my primary concern, but also the complications that will come from our dependencies forcing our hand. I guess when a dependency (say numba) fixes an issue in a new release, and breaks compatibility with old-Python, we will have to find workarounds to keep supporting the old numba version in some way. Possibly this means we have to keep a slower code path around for old-Python.

Two to three versions with full support would be great. Once 3.9 is supported, we would fully support 3.7 through 3.9, until approx. mid 2023, [when upstream support for 3.7 is dropped](https://www.python.org/dev/peps/pep-0537/#lifespan). Then, 3.7 will be treated like we will [soon treat 3.6](https://www.python.org/dev/peps/pep-0494/#lifespan), maybe keeping 3.6 if it is still in use.

About the testing regime, I'm not so sure. Only running such a small subset of tests that it finished quickly could result in quality issues down the line, if some kind of manual triggering is needed for running with more complete coverage. Maybe we should formulate our goals, instead of starting with a method. With the CI matrix for "fully supported" Python versions, the goal is to have an always stable master branch, which means that development is never stalled or hindered by bugs introduced in PRs, and that continuous releases pass the test suite. The stability extends to full releases, of course.

For "legacy support" versions, I think we should concentrate on the high-quality releases, while not being a burden on the CI performance, which is always also a burden on the contributors. There are a few strategies for reducing this burden:

1) Don't run the tests for legacy versions continuously, run them with limited frequency (for example nightly, weekly etc., not for each push). Runs are automatic, but someone has to take care of failures.
2) Take some shortcuts. For example, it could be possible to cache the full tox environments between runs, so we won't always test on current versions of dependencies, and run the "uncached" pipeline only manually before releases. Again, manually running can be problematic.
3) Just throw more resources at the problem. Combine multiple CI platforms again, and keep "fully supported" Python versions in the continuous path with the release scripts. If we choose to use both GitHub Actions and Azure Pipelines, we actually can [trigger a pipeline from an action](https://docs.microsoft.com/en-us/azure/devops/pipelines/ecosystems/github-actions?view=azure-devops), which could enable us to keep both in the continuous path.

Let me know if this makes sense, or if you have any additional ideas.

uellue, 2020-12-15:
> Let me know if this makes sense, or if you have any additional ideas.

Yes, this all makes a lot of sense. :-) Good point to focus on the intended outcome and not the path to get there!

Ok, so there are two things to consider in development: 

1. Make sure we don't _unnecessarily_ introduce language constructs or dependencies on 3rd party software that break support for older Python versions.
2. If using such a new feature brings significant benefits, consider introducing code paths that handle the situation gracefully on older Python versions, such as a helpful error message or an alternative fallback implementation -- say, "un-shared" memory, uncached JIT, ...

In both cases we do need CI to inform us of an issue as soon as possible in the development process since designing it properly for backwards compatibility from the start can be easier than fixing it later. That means older versions should be tested with each PR and push. Maybe one should even test _primarily_ against the oldest and the newest supported version with cached TOX in PRs, and test other versions daily? It would be great if the CI environment was more stable so that upstream changes don't disrupt development, and upstream changes would be caught in a daily run. Maybe even automatically opening an Issue if a daily build fails? :-)

sk1p, 2021-01-14:
numba now supports 3.9 in the latest development version

sk1p, 2021-03-17:
Numba 0.53 was recently released, so now we are only waiting on the dask issue. It's possible that it's actually a Python upstream issue, so we may want to implement a work-around if we want to release 0.7 with Python 3.9 support (which would be wonderful!)

sk1p, 2021-04-14:
For the dask issue, we can do something like this: https://github.com/usnistgov/cord19-cdcs-nist/commit/d4aec8d606e084a72506de995ce02456d6229b16

uellue, 2021-06-01:
It seems to work now. I've installed it in a fresh 3.9 conda environment. GUI works, and the regular test pass. I'm updating the install information accordingly.

We could now think about activating CI for 3.9. :-)

PR #1037 (2021-06-01): Support Python 3.9
Closes #914

Documentation and changelog will be handled in #1034 

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the client](https://libertem.github.io/LiberTEM/co [... cut]
commit 4b143114 (2021-06-01): Support Python 3.9 (#1037)

Closes #914

* Build distributed pipeline image on top of a Python Docker image, not stock Debian
* Run data pipeline on Python 3.6, 3.7 and 3.9
* Update Azure agent on moellenstedt and allow multiple Python versions (not in this repo yet)
* Use unique name to distinguish pipelines in analytics
* Exclude bad Python 3.9
closed 2021-06-01 by uellue