LiberTEM/LiberTEM #917: Escalating memory use in point analysis on RAW or EMPAD file

uellue, 2020-12-07:
There seems to be a memory leak or a different mechanism where LiberTEM workers consume excessive amounts of memory when doing a point analysis.

How to reproduce:

* Current master branch
* Windows 10 
* Web GUI
* Open RAW pr EMPAD file (tested with BiFeO3 scan_11_x256_y256.raw)
* Run a point analysis

What happens:

```
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.90 GB -- Worker memory limit: 2.07 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.91 GB -- Worker memory limit: 2.07 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.91 GB -- Worker memory limit: 2.07 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.91 GB -- Worker memory limit: 2.07 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.92 GB -- Worker memory limit: 2.07 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.92 GB -- Worker memory limit: 2.07 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 1.95 GB -- Worker memory limit: 2.07 GB
distributed.nanny - WARNING - Worker exceeded 95% memory budget. Restarting
distributed.nanny - WARNING - Restarting worker
```

After that, LiberTEM is locked up and the server has to be restarted.

## Differential diagnosis

* Does not occur with the initial sum analysis
* Does not occur with ring or disk analysis
* Does not occur in last release (v0.5.0)
* Does not occur with same data in EMD format (HDF5)
* Does not occur with a MIB file of comparable size
* Does occur also when opening file as EMPAD

## Installed packages

<details>

```
python==3.7.4

aio-pika==6.6.1
aiohttp==3.6.2
aiormq==3.2.2
alabaster==0.7.12
apipkg==1.5
appdirs==1.4.4
argh==0.26.2
asv==0.4.2
async-generator==1.10
async-timeout==3.0.1
asyncio==3.4.3
atomicwrites==1.3.0
attrs==19.2.0
autoflake==1.4
autopep8==1.5.4
Babel==2.7.0
backcall==0.1.0
bibtexparser==1.1.0
black==20.8b1
bleach==3.1.0
certifi==2019.9.11
chardet==3.0.4
click==7.1.2
cloudpickle==1.5.0
cmake==3.17.0
colorama==0.4.1
coverage==5.0.4
cycler==0.10.0
Cython==0.29.13
dask==2.21.0
decorator==4.4.0
defusedxml==0.6.0
Deprecated==1.2.6
deprecation==2.1.0
dill==0.3.1.1
distributed==2.21.0
docutils==0.15.2
empyre==0.3.0
entrypoints==0.3
execnet==1.7.1
fastrlock==0.4
filelock==3.0.12
flake8==3.7.8
future==0.18.0
ghp-import==0.5.5
graphviz==0.13.2
h5py==2.10.0
h5py-cache==1.0
hdbscan==0.8.23
hdfs3==0.3.1
HeapDict==1.0.1
html5lib==1.0.1
hyperspy==1.5.2
idna==2.8
imageio==2.6.1
imagesize==1.1.0
importlib-metadata==1.2.0
ipykernel==5.3.4
ipyparallel==6.2.4
ipython==7.17.0
ipython-genutils==0.2.0
ipywidgets==7.5.1
jedi==0.15.1
Jinja2==2.10.3
joblib==0.14.0
jsonschema==3.1.1
jupyter==1.0.0
jupyter-client==5.3.4
jupyter-console==6.0.0
jupyter-core==4.6.1
kiwipy==0.6.0
kiwisolver==1.1.0
latexcodec==1.0.7
-e git+https://github.com/uellue/LiberTEM.git@8203ae22c2ce91c0a7e45e3ca4968f84e9bc3cbf#egg=libertem
libertem-blobfinder==0.4.1
-e git+https://github.com/uellue/libertem-hackaton@23415310fbc77e5cb15c979fc58c4cd25b3d0651#egg=libertem_hackaton
line-profiler==3.0.2
livereload==2.6.1
llvmlite==0.34.0
MarkupSafe==1.1.1
matplotlib==3.1.1
mccabe==0.6.1
mistune==0.8.4
more-itertools==7.2.0
mpmath==1.1.0
msgpack==0.6.2
multidict==4.5.2
mypy-extensions==0.4.3
natsort==7.0.1
nbconvert==5.6.0
nbformat==4.4.0
nbsphinx==0.7.1
nbsphinx-link==1.3.0
ncempy==1.5.0
networkx==2.3
-e git+https://github.com/uellue/niondata.git@57aa91493a68dd8b0e2ebb66529ad3005bf5900f#egg=niondata
-e git+https://github.com/uellue/nionswift.git@e90df42bb607c3b4861a4fc8a820b712881b6953#egg=nionswift
-e git+https://github.com/nion-software/nionswift-io.git@0bb981aafddc2875660f88576313409f957f46df#egg=nionswift_io
nionswift-tool==0.3.25
-e git+https://github.com/nion-software/nionui.git@340cf256b0b6455c1b865cd143a0778a4ee17a6f#egg=nionui
-e git+https://github.com/nion-software/nionutils.git@7214bd8559a871c46e379ffbc0b27f115ad11b44#egg=nionutils
notebook==6.0.1
numba==0.51.2
numexpr==2.7.1
numpy==1.19.0
oset==0.1.3
packaging==19.2
pamqp==2.3.0
pandas==1.0.0
pandocfilters==1.4.2
parso==0.5.1
pathspec==0.8.0
pathtools==0.1.2
patsy==0.5.1
pickleshare==0.7.5
pika==1.1.0
Pillow==6.2.0
Pint==0.10.1
pluggy==0.13.0
plumbum==1.6.9
port-for==0.3.1
primesieve==2.2.0
prometheus-client==0.7.1
prompt-toolkit==2.0.10
psutil==5.6.3
PTable==0.9.2
py==1.8.0
py-cpuinfo==6.0.0
pybind11==2.5.0
pybtex==0.22.2
pybtex-docutils==0.2.2
pycodestyle==2.6.0
pyflakes==2.1.1
pygal==2.4.0
pygaljs==1.0.2
PyGithub==1.43.8
Pygments==2.4.2
PyJWT==1.7.1
pypandoc==1.5
pyparsing==2.4.2
pyrsistent==0.15.4
pytest==5.4.3
pytest-asyncio==0.14.0
pytest-benchmark==3.2.3
pytest-cov==2.10.0
pytest-forked==1.1.3
pytest-sphinx==0.2.2
pytest-xdist==1.34.0
python-dateutil==2.8.0
pytray==0.2.2
pytz==2019.3
PyWavelets==1.0.3
pywin32==227
pywinpty==0.5.5
PyYAML==5.1.2
pyzmq==18.1.0
qtconsole==4.5.5
regex==2020.9.27
requests==2.22.0
rope==0.16.0
rpyc==4.1.5
scikit-image==0.16.1
scikit-learn==0.21.3
scipy==1.4.1
Send2Trash==1.5.0
shortuuid==1.0.1
six==1.14.0
snowballstemmer==2.0.0
sortedcontainers==2.1.0
sparse==0.8.0
Sphinx==3.2.0
sphinx-autobuild==0.7.1
sphinx-issues==1.2.0
sphinxcontrib-applehelp==1.0.1
sphinxcontrib-bibtex==1.0.0
sphinxcontrib-devhelp==1.0.1
sphinxcontrib-htmlhelp==1.0.2
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.2
sphinxcontrib-serializinghtml==1.1.3
statsmodels==0.11.0
sympy==1.5.1
tblib==1.6.0
terminado==0.8.2
testfixtures==6.11.0
testpath==0.4.2
threadpoolctl==2.0.0
toml==0.10.1
toolz==0.10.0
torch==1.3.0
tornado==6.0.3
tox==3.14.6
tqdm==4.42.0
traitlets==4.3.3
traits==5.2.0
travis-sphinx==2.2.1
typed-ast==1.4.1
typing-extensions==3.7.4.3
tzlocal==2.0.0
urllib3==1.25.6
virtualenv==16.7.5
watchdog==0.10.2
wcwidth==0.1.7
webencodings==0.5.1
websockets==8.1
widgetsnbextension==3.5.1
wincertstore==0.2
wrapt==1.11.2
yarl==1.3.0
zict==1.0.0
zipp==0.6.0
```
</details>

uellue, 2020-12-07:
The issue is reproducible on Linux (moellenstedt) on /cachedata/users/clausen/tiled.raw. The memory consumption of workers increases dramatically from 0.1 % to almost 1 % of the available memory while a point analysis is running. Since this system has a very large main memory, this doesn't lead to failure. Furthermore, the analysis is very slow compared to ring analysis. The memory use drops back to normal once the analysis is finished.

Could this be caused by the sparse dot product copying the tile internally? I'll try setting tile size hints.

uellue, 2020-12-07:
Restricting the tile size in ApplyMasksUDF with 

```
    def get_tiling_preferences(self):
        """
        Configure tiling preferences. Return a dictinary with the
        following keys:

        - "depth": number of frames/frame parts to stack on top of each other
        - "total_size": total size of a tile in bytes
        """
        return {
            "depth": UDF.TILE_DEPTH_DEFAULT,
            "total_size": 1e6,
        }
```

resolves the issue. However, the sparse dot product shouldn't consume so much memory and it used to be faster and not slower.  SInce this is a performance-critical part of LiberTEM, probably the root cause should be found instead of just patching it up?

sk1p, 2020-12-07:
Interesting, thanks for the report. Note that with Python, most of the time high memory usage is not really a leak, but rather high GC pressure. This can be tested by repeating the test: if there is more or less monotone increase in memory use, there is a real leak, otherwise, it's possible that it is "only" excessive garbage. 

I can reproduce this problem here, with memory usage shooting up to 1GiB per worker and the analysis taking much more time than other mask-based analyses. As the data is read straight from the memory map, we have a tile depth of `9362`, which results in tiles of almost 600MiB. After the run, memory usage goes back down to ~200MiB per worker. The sparse dot creating a copy would fully explain the behavior, I think:

> Does not occur with the initial sum analysis

The sum doesn't copy its argument

> Does not occur with ring or disk analysis

Ring/disk are dense; the dense dot product doesn't copy data around

> Does not occur in last release (v0.5.0)

As far as I remember, 0.5 didn't have the optimization of reading large tiles if the data was read straight via mmap

> Does not occur with same data in EMD format (HDF5)
> Does not occur with a MIB file of comparable size

HDF5 and MIB don't read data with large mmap'd tiles

> Does occur also when opening file as EMPAD

RAW and EMPAD use the same code paths w.r.t. tiling

uellue, 2020-12-07:
> The sparse dot creating a copy would fully explain the behavior, I think:

Ok, I'll try to build a minimal reproducer. :-)

sk1p, 2020-12-07:
Another data point for the diagnosis: try to run a radial fourier analysis with a very small ring. On my system that reproduces the problem, compared to a larger ring which runs in acceptable time and with acceptable memory usage.

> After that, LiberTEM is locked up and the server has to be restarted.

I think this could be related to the [handling of OOM situations by dask](https://distributed.dask.org/en/latest/worker.html#memory-management) - according to the messages you posted above, it's restarting the worker at 95% memory usage. But for some reason, LiberTEM hangs in that situation - it's not clear to me why that is (Is the worker properly restarted? Do we need to do anything? I assumed that things like restarting jobs are abstracted away from us by the distributed future interface)

uellue, 2020-12-07:
Yes, the problem is with the sparse dot product. If I memory-map a large raw file and multiply it with the sparse matrix using '@', the memory use balloons.

sk1p, 2020-12-07:
So this means that `flat_data @ masks` actually internally calls the `dot` method on the ndarray, which converts its argument to a dense array? Do we need to always use the `masks.dot(...)` method in case of sparse masks? I.e. first branch here:

https://github.com/LiberTEM/LiberTEM/blob/8203ae22c2ce91c0a7e45e3ca4968f84e9bc3cbf/src/libertem/udf/masks.py#L156-L161

uellue, 2020-12-07:
The '@' operator calls internally the `__rmatmul__` and `__matmul__` methods. To exclude any issues with the "routing" and make sure that the correct implementation is used, here's an example that uses these methods directly.

This is really striking since I remember to have tested and optimized this in the past. However, I couldn't resolve it by installing older versions of numpy and scipy. I'm a bit lost regarding what to do now. Transposing the data leads to a really funny effect, by the way: The right hand side gets a lot faster, and the left hand side becomes as slow as the right hand side. It looks like the implementation is lazy and just internally transposes the data. That is mathematically correct, but kills performance.

```python
import numpy as np
import scipy.sparse

data = np.zeros((16384, 16384), dtype=np.float32)
sparse_mask_left = scipy.sparse.csr_matrix(([1.], ([0], [0])), shape=(16384, 1), dtype=np.float32)

%timeit sparse_mask_left.__rmatmul__(data)
```

    846 ms ± 2.14 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

```python
%timeit sparse_mask_left.__rmatmul__(data.T)
```

    70.5 µs ± 489 ns per loop (mean ± std. dev. of 7 runs, 10000 loops each)

```python

dense_mask_left = sparse_mask_left.toarray()

%timeit dense_mask_left.__rmatmul__(data)
```

    2.32 ms ± 68.7 µs per loop (mean ± std. dev. of 7 runs, 100 loops each)

```python
sparse_mask_right = scipy.sparse.csr_matrix(([1.], ([0], [0])), shape=(1, 16384), dtype=np.float32)

%timeit sparse_mask_right.__matmul__(data)
```

    10 µs ± 31.6 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)

```python
%timeit sparse_mask_right.__matmul__(data.T)
```

    841 ms ± 2.39 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

```python
dense_mask_right = sparse_mask_right.toarray()

%timeit dense_mask_right.__matmul__(data)
```

    3.87 ms ± 102 µs per loop (mean ± std. dev. of 7 runs, 100 loops each)

uellue, 2020-12-07:
Maybe one could break out or own Numba version of CSR and CSC dot products to resolve this?

Also, this may have implications for SSB in ptychography. Not sure why it didn't occur there.

PR #920 (2020-12-08): Custom right hand side matrix multiplication
...for scipy.sparse.{csc,csr]_matrix

Closes #917
See also https://github.com/scipy/scipy/issues/13211

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases

## Reviewer Checklist:

* [x] `/azp run libe [... cut]
commit 55dcc90b (2020-12-08): Custom right hand side matrix multiplication

...for scipy.sparse.{csc,csr]_matrix

Closes #917
See also https://github.com/scipy/scipy/issues/13211
closed 2020-12-08 by sk1p