LiberTEM/LiberTEM #925: Visualization: remove outliers before normalization

sk1p, 2020-12-14:
If there are unusually large outliers, the default visualization is significantly impacted, as they throw off the normalization:

[image]

Something like [these functions in libertem-holo](https://github.com/LiberTEM/LiberTEM-holo/blob/43363ebdb749ec3e792c58ada98e723d09ec4070/src/libertem_holo/base/filters.py#L14-L31) could be useful.

uellue, 2020-12-14:
The issue comes from negative values in combination with log scaling. Since negative values can't be represented well in a log scaled plot, they are shown as black by matplotlib.

uellue, 2020-12-14:
Discussion with @sk1p: My bad, this was on the right where we have linear scaling. It is in fact an outlier that drags all other values to the very bottom of the color scale.

sk1p, 2020-12-14:
Interim solution: for virtual detector analyses, in addition to linear scale output, also output a layer with a log-scaled visualization. As opposed to just patching out the outliers, this would allow to view data with high dynamic range without loss of information.

Medium-term: allow to set normalization parameters in client. Has about four steps:
- get vmin/vmax information to the client
- visualization update endpoint (could be a simple "synchronous" GET request that returns the visualization result for given parameters)
- update state management to allow to update visualization results
- add GUI elements to change visualization parameters

Long-term: #134

PR #929 (2020-12-16): Allow choosing between lin and log visualization
Closes #925 with a small modification to make sure large outliers
don't generate a completely blank plot.

See https://github.com/LiberTEM/LiberTEM/issues/925#issuecomment-744483882
for more discussions and an idea for a more comprehensive solution

Full implementation would be #134

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/sourc [... cut]
commit 9a025b0f (2020-12-16): Allow choosing between lin and log visualization

Closes #925 with a small modification to make sure large outliers
don't generate a completely blank plot.

See https://github.com/LiberTEM/LiberTEM/issues/925#issuecomment-744483882
for more discussions and an indea for a more comprehensive solution

Full implementation would be #134
closed 2020-12-16 by sk1p