LiberTEM/LiberTEM #928: Documentation, test case and example for UDFMeta.coordinates

uellue, 2020-12-15:
As a follow-up to #793, we should refine the newly introduced feature. A good candidate to use this in practice is SSB in https://github.com/Ptychography-4-0/ptychography

uellue, 2021-03-18:
In use in https://github.com/Ptychography-4-0/ptychography/, can serve as an example

AnandBaburajan, 2021-03-25:
> In use in https://github.com/Ptychography-4-0/ptychography/, can serve as an example

I've written some tests [here](https://github.com/LiberTEM/LiberTEM/blob/master/tests/udf/test_coords.py) and little documentation [here](https://libertem.github.io/LiberTEM/reference/udf.html#libertem.udf.base.UDFMeta.coordinates). I'm not sure if I should write application-specific (would need to understand ptychography?) documentation and examples.

sk1p, 2021-03-25:
Hmm, I think it could be useful to also have a paragraph or two in the non-reference documentation, maybe as part of the ["advanced UDFs" section](https://libertem.github.io/LiberTEM/udf/advanced.html#meta-information)? Doesn't have to be a complex example - maybe just informing the user that the coords are there.

@uellue can you think of a simpler example that approximates something that would be used in the wild? Maybe a simple example of descan correction?

PR #1034 (2021-06-09): Docs update for 0.7
Closes #990
Closes #928

## TODO

- [x] Add changelog entry for #1039

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [N/A] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [x] I have included the [rebuilt production build of the client](https://libertem.github.io/Liber [... cut]
commit 3d9320fb (2021-06-09): Restructure and extend UDF docs; update README

* Make advanced UDF topics findable with a better TOC
* Reduce file size by splitting into intro and basic UDF usage
* Document coordinates, closes #928
* Document UDF constructors
* Explanation of MapReduce in README
closed 2021-06-09 by sk1p