LiberTEM/LiberTEM #930: AppImage: restarting workers

sk1p, 2020-12-16:
The continuous AppImage somehow has trouble with keeping the distributed workers alive. Browsing for file still works, but when detecting parameters, workers are restarted:

```
distributed.nanny - WARNING - Restarting worker
distributed.nanny - WARNING - Restarting worker
distributed.nanny - WARNING - Restarting worker
tornado.application - ERROR - Uncaught exception GET /api/datasets/detect/?path=%2Fhome%2Falex%2Fsource%2FLiberTEM%2Fdata%2FEMPAD%2Fscan_11_x256_y256.raw (::1)
HTTPServerRequest(protocol='http', host='localhost:9000', method='GET', uri='/api/datasets/detect/?path=%2Fhome%2Falex%2Fsource%2FLiberTEM%2Fdata%2FEMPAD%2Fscan_11_x256_y256.raw', version='HTTP/1.1', remote_ip='::1')
Traceback (most recent call last):
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/site-packages/tornado/web.py", line 1704, in _execute
    result = await result
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/site-packages/libertem/web/dataset.py", line 135, in get
    detected_params = await run_blocking(
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/site-packages/libertem/utils/async_utils.py", line 18, in run_blocking
    return await tornado.ioloop.IOLoop.current().run_in_executor(None, partial(fn, *args, **kwargs))
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/site-packages/libertem/io/dataset/__init__.py", line 113, in detect
    params = cls.detect_params(path, executor)
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/site-packages/libertem/io/dataset/hdf5.py", line 160, in detect_params
    executor.run_function(cls._do_detect, path)
  File "/tmp/.mount_LiberTQd6L8j/usr/lib/python3.8/site-packages/libertem/executor/dask.py", line 320, in run_function
    return future.result()
  File "/home/alex/.local/lib/python3.8/site-packages/distributed/client.py", line 223, in result
    raise exc.with_traceback(tb)
distributed.scheduler.KilledWorker: ('_do_detect-315698e267c6202779569ab2981d50f4', <Worker 'tcp://192.168.178.27:39491', name: default-cpu-3, memory: 0, processing: 1>)
distributed.nanny - WARNING - Restarting worker
```

This was reproduced with `LiberTEM-791dea21-x86_64.AppImage` from the [continuous build](https://github.com/LiberTEM/LiberTEM/releases/tag/continuous)

sk1p, 2020-12-16:
This is possibly related to my local pyenv setup - this part is suspicious:

```
  File "/home/alex/.local/lib/python3.8/site-packages/distributed/client.py", line 223, in result
    raise exc.with_traceback(tb)
```

uellue, 2020-12-16:
The AppImage seems to work fine on Moellenstedt, which would support the hypothesis that it is related to your setup @sk1p. Can we somehow make the AppImage more robust against that? :-)

sk1p, 2021-04-23:
Now that I'm no longer using pyenv, the AppImage also works for me again. I don't know if the AppImage can be made more robust in case of pyenv being used. I think the clear workaround is not using the AppImage, but installing LiberTEM in a venv as usual. I'm also thinking that the overlap of pyenv users and LiberTEM users that use the AppImage for easy installation should be quite small, otherwise we would have gotten reports about this already.

closed 2021-04-23 by sk1p