LiberTEM/LiberTEM #937: Validate file readers against alternative implementations

uellue, 2021-01-22:
In #932, some tests that use a form of snapshot testing are replaced with tests that validate against alternative file reader implementations to make the results more stable against numerical variations and consistent with other relevant packages.

A prototype notebook with alternative readers for all  formats except K2IS is included in that PR. However, only selected tests for DM, FRMS6 and MRC are converted yet which were affected by the numerical instability. The alternative readers should be included for other file format tests as well:

- [ ] FRMS6 full file, not only beginning
- [ ] DM full file and 3D DM
- [ ] MRC full file
- [ ] BLO
- [ ] EMPAD
- [ ] MIB
- [ ] SEQ
- [ ] SER

uellue, 2021-01-23:
Should we perhaps include a raw data version of K2IS generated with the 0.5 release as a snapshot and reference? At least we will catch changes!

sk1p, 2021-01-23:
> Should we perhaps include a raw data version of K2IS generated with the 0.5 release as a snapshot and reference? At least we will catch changes!

Sounds like a plan! :+1:

uellue, 2021-01-26:
Other formats are covered as well now, this issue will be closed with #932.

PR #932 (2021-01-26): Improve file format test stability; use process_tile() in SumSigUDF
Make sure that file format tests are stable against numerical variations in the UDFs that generate reference results. Validate with `allclose()` against results that are generated with an alternative reader and calculation method instead of comparing exact snapshots.

Closes #937 

Original reason: `process_tile()` in `SumSigUDF` is significantly faster, about 1.5x better in some tests

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master [... cut]
closed 2021-01-26 by sk1p