LiberTEM/LiberTEM #940: Windows: number of workers mismatch in CI

sk1p, 2021-01-25:
`test_local_cluster_details` fails with the following error:

```
2021-01-25T18:17:37.8445259Z E           AssertionError: assert {'details': [{'cpu': 5, 'cuda': 0, 'host': '192.168.0.103', 'service': 1}],\n 'messageType': 'CLUSTER_DETAILS',\n 'status': 'ok'} == {'details': [{'cpu': 7, 'cuda': 0, 'host': '192.168.0.103', 'service': 1}],\n 'messageType': 'CLUSTER_DETAILS',\n 'status': 'ok'}
2021-01-25T18:17:37.8446404Z E             Common items:
2021-01-25T18:17:37.8446878Z E             {'messageType': 'CLUSTER_DETAILS', 'status': 'ok'}
2021-01-25T18:17:37.8447327Z E             Differing items:
2021-01-25T18:17:37.8448112Z E             {'details': [{'cpu': 5, 'cuda': 0, 'host': '192.168.0.103', 'service': 1}]} != {'details': [{'cpu': 7, 'cuda': 0, 'host': '192.168.0.103', 'service': 1}]}
2021-01-25T18:17:37.8448895Z E             Full diff:
2021-01-25T18:17:37.8449188Z E               {
2021-01-25T18:17:37.8449685Z E             -  'details': [{'cpu': 7, 'cuda': 0, 'host': '192.168.0.103', 'service': 1}],
2021-01-25T18:17:37.8450593Z E             ?                      ^
2021-01-25T18:17:37.8451153Z E             +  'details': [{'cpu': 5, 'cuda': 0, 'host': '192.168.0.103', 'service': 1}],
2021-01-25T18:17:37.8451672Z E             ?                      ^
2021-01-25T18:17:37.8452086Z E                'messageType': 'CLUSTER_DETAILS',
2021-01-25T18:17:37.8454380Z E                'status': 'ok',
2021-01-25T18:17:37.8454806Z E               }
2021-01-25T18:17:37.8455021Z 
```

And logs the following:

```
2021-01-25T18:13:49.3254820Z tests/server/test_cluster_status.py::test_local_cluster_details tornado.application - ERROR - Exception in callback <bound method Nanny.memory_monitor of <Nanny: None, threads: 1>>
2021-01-25T18:13:49.3255860Z Traceback (most recent call last):
2021-01-25T18:13:49.3256363Z   File "d:\a\1\s\.tox\py38\lib\site-packages\tornado\ioloop.py", line 905, in _run
2021-01-25T18:13:49.3256802Z     return self.callback()
2021-01-25T18:13:49.3257254Z   File "d:\a\1\s\.tox\py38\lib\site-packages\distributed\nanny.py", line 414, in memory_monitor
2021-01-25T18:13:49.3257741Z     process = self.process.process
2021-01-25T18:13:49.3258057Z AttributeError: 'NoneType' object has no attribute 'process'
2021-01-25T18:13:49.3482178Z tornado.application - ERROR - Exception in callback <bound method Nanny.memory_monitor of <Nanny: None, threads: 1>>
2021-01-25T18:13:49.3483789Z Traceback (most recent call last):
2021-01-25T18:13:49.3484524Z   File "d:\a\1\s\.tox\py38\lib\site-packages\tornado\ioloop.py", line 905, in _run
2021-01-25T18:13:49.3485169Z     return self.callback()
2021-01-25T18:13:49.3485491Z   File "d:\a\1\s\.tox\py38\lib\site-packages\distributed\nanny.py", line 414, in memory_monitor
2021-01-25T18:13:49.3485838Z     process = self.process.process
2021-01-25T18:13:49.3486107Z AttributeError: 'NoneType' object has no attribute 'process'
2021-01-25T18:13:49.4176987Z tornado.application - ERROR - Exception in callback <bound method Nanny.memory_monitor of <Nanny: None, threads: 1>>
[repeated N times]
```

Maybe the `AttributeError`s are not the problem, but just a symptom of the cluster being shutdown before it had the chance to actually start the workers? In the back in my mind I had the idea that we wait for the workers to be ready when starting a local cluster, but it seems we only do this for the `cli_worker`:

https://github.com/LiberTEM/LiberTEM/blob/ece9a6e7626ae88024f655245036a526f88599e1/src/libertem/executor/dask.py#L491-L493

Maybe we have to do something like that in `make_local` too? Call [`Client.wait_for_workers`](https://distributed.dask.org/en/latest/api.html#distributed.Client.wait_for_workers)? In addition, we could reduce the maximum number of workers in that specific test case.

uellue, 2021-01-25:
I couldn't reproduce this issue on my workstation FYI. I ran the offending test, the server tests and the whole test suite repeatedly.

sk1p, 2021-01-25:
> I couldn't reproduce this issue on my workstation FYI. I ran the offending test, the server tests and the whole test suite repeatedly.

Was that with a fresh environment? There was a distributed release a few days ago, that may be related.

sk1p, 2021-01-27:
Should be fixed by #941

closed 2021-01-27 by sk1p