LiberTEM/LiberTEM #942: Release 0.6

sk1p, 2021-01-26:
# Release checklist

## Remaining issues

#523 #827 #892 #939 (see [0.6 milestone](https://github.com/LiberTEM/LiberTEM/issues?q=is%3Aopen+milestone%3A0.6+sort%3Aupdated-desc))

## Before (using a release candidate package)

- [ ] Review open issues and pull requests
- [ ] Handle deprecation, search the code base for DeprecationWarning that are supposed to be removed in that release.
- [ ] Full documentation review and update, including link check using sphinx-build -b linkcheck "docs/source" "docs/build/html"
- [ ] Run complete test suite, including slow tests that are deactivated by default and tests that require sample files.
- [ ] Update the expected version in notes on changes, i.e. from 0.3.0.dev0 to 0.3.0 when releasing version 0.3.0.
- [ ] Update and review change log in docs/source/changelog.rst, merging snippets in docs/source/changelog/*/ as appropriate.
- [ ] Update the JSON files in the packaging/ folder with author and project information (see also #827)
- [ ] Edit setup.cfg to exclude flaky tests temporarily from release builds
- [ ] Create a release candidate using scripts/release. See scripts/release --help for details.
- [ ] Confirm that wheel, tar.gz, and AppImage are built for the release candidate on GitHub
- [ ] Confirm that a new version with the most recent release candidate is created in the Zenodo.org sandbox that is ready for submission.
- [ ] Install release candidate packages in a clean environment (for example: pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple 'libertem==0.6.0rc1')
- [ ] For the GUI-related items, open in an incognito window to start from a clean slate
- [ ] Correct version info displayed in info dialogue?
- [ ] Link check in version info dialogue
- [ ] Make sure you have test files of all supported types available
  - [ ] Include floats, ints, big endian, little endian, complex raw data
- [ ] Open each test file
  - [ ] Are parameters recognized correctly, as far as implemented?
  - [ ] Any bad default values?
  - [ ] Does the file open correctly?
  - [ ] Have a look at the dataset info dialogue. Reasonable values?
- [ ] Perform all analyses on each test file.
  - [ ] Does the result change when the input parameters are changed?
  - [ ] All display channels present and looking reasonable?
  - [ ] Reasonable performance?
  - [ ] Use pick mode.
- [ ] Re-open all the files
  - [ ] Are the files listed in “recent files”?
  - [ ] Are the parameters filled from the cache correctly?
- [ ] Try opening all file types with wrong parameters
  - [ ] Proper understandable error messages?
- [ ] Pick one file and confirm keyboard and mouse interaction for all analyses
  - [ ] Correct bounds check for keyboard and mouse?
-[ ] Check what happens when trying to open non-existent files or directories in the GUI.
  - [ ] Proper understandable error message?
  - [ ] Possible to continue working?
- [ ] Shut down libertem-server while analysis is running
  - [ ]  Shut down within a few seconds?
  - [ ] All workers reaped?
- [ ] Check what happens when trying to open non-existent files by scripting.
  - [ ] Proper understandable error message? TODO automate?
- [ ] Check what happens when opening all file types with bad parameters by scripting
  - [ ] Proper understandable error message? TODO automate?
- [ ] Run all examples
- [ ] Run libertem-server on Windows, connect to a remote dask cluster running on Linux, open all file types and perform an analysis for each file type.
- [ ] Use the GUI while a long-running analysis is running
  - [ ] Still usable, decent response times?

- [ ] Confirm that pull requests and issues are handled as intended, i.e. milestoned and merged in appropriate branch.

## After releasing on GitHub
- [ ] Confirm that all release packages are built and release notes are up-to-date
- [ ] Install release package
- [ ] Confirm correct version info
- [ ] confirm package upload to PyPi
- [ ] Publish new version on zenodo.org
- [ ] Update documentation with new links, if necessary
  - [ ] Add zenodo badge for the new release to Changelog page
- [ ] Send announcement message on mailing list
- [ ] Edit setup.cfg to include flaky tests again
- [ ] Bump version in master branch to next .dev0
- [ ] Add the current LiberTEM version to CVL (add both the singularity and the .desktop file!)

sk1p, 2021-01-27:
Just as a note, 0.6.0rc0 did not finish, as there were two problems:

1) `nest-asyncio` did a release 1.5.0 just before us, which was broken. They quickly followed up with 1.5.1, which fixed the problem
2) The username passed on to `twine` was wrong, it needs to be `__token__`, not `libertem_bot`. This should now be fixed for `0.6.0rc1`

sk1p, 2021-01-27:
Zenodo tokens are also still missing @uellue - can we fix that tomorrow morning? rc1 is available for testing, so we can continue anyways.

uellue, 2021-02-05:
FYI, I started going through the reference data and made it until NeXus. More next week...

sk1p, 2021-02-09:
We should probably also make sure that 0.6 works with the next numba release:

https://numba.discourse.group/t/numba-0-53-0-and-llvmlite-0-36-0-release-candidates/504

sk1p, 2021-02-11:
> We should probably also make sure that 0.6 works with the next numba release:
> 
> https://numba.discourse.group/t/numba-0-53-0-and-llvmlite-0-36-0-release-candidates/504

Test suite passes, cursory usage tests looked fine :+1:

sk1p, 2021-02-16:
CVL update submitted: https://github.com/Characterisation-Virtual-Laboratory/CharacterisationVL-Software/pull/115

uellue, 2021-02-16:
@AnandBaburajan @twentyse7en your contributions are now finally released with version 0.6! Both reshaping and templates are great features that are among the highlights of this release. 🚀  Thanks again!

uellue, 2021-02-16:
Post release checklist completed, closing! 🍾

closed 2021-02-16 by uellue

AnandBaburajan, 2021-02-16:
> @AnandBaburajan @twentyse7en your contributions are now finally released with version 0.6! Both reshaping and templates are great features that are among the highlights of this release.  Thanks again!

Thanks to you and @sk1p for the opportunity! I'll get back to my pending issues once I'm done with my exams and job-hunt! Looking forward to keep contributing to LiberTEM!