LiberTEM/LiberTEM #944: Visualization: linear viz doesn't properly show binary images

sk1p, 2021-01-28:
LiberTEM version: 0.5.1 or later (didn't test earlier)

Test code:

```python
from libertem.viz import visualize_simple
import matplotlib.pyplot as plt
import numpy as np

my_choice = np.random.choice(size=(128, 128), a=[0, 1], p=[0.8, 0.2])
rgb_data = visualize_simple(my_choice)
plt.figure()
plt.imshow(rgb_data)
```

Should show some sparse 10% random 1/0 noise, all-black instead.

Probably a one-off error, compare with result of this:

```python
my_choice = np.random.choice(size=(128, 128), a=[0, 1, 2], p=[0.8, 0.1, 0.1])
rgb_data = visualize_simple(my_choice)
plt.figure()
plt.imshow(rgb_data)
```

Another way to reproduce: point analysis on 1bit data set.

sk1p, 2021-04-14:
Will hopefully be fixed in #1011

uellue, 2021-04-15:
Fix confirmed with a boolean test dataset

PR #1011 (2021-04-22): `Context.run_udfs` improvements
Fixes #944 
Fixes #1016
Fixes #980 

# Multiple improvements in `Context.run_udf`

* Allow running multiple UDFs "at once" on a single `DataSet`
* Allow usage from an asynchronous context
* Allow getting results iteratively as a generator (both sync and async)
* Simple plotting, including live-updates (`Context.run_udf(..., plots=True)` for very simple usage)

## Under the hood:

* Add `libertem.viz.{mpl,bgp, gms}` packages, which implement live-updated plotting
* remove code duplication in `UDFRunner.run_for_dataset{,_async}`
* call `UDFRunner.run_for_dataset_s [... cut]
closed 2021-04-22 by sk1p