LiberTEM/LiberTEM #946: GPU: I/O and performance optimizations

sk1p, 2021-02-01:
GPU support is still experimental, and needs to be looked at from a performance perspective. Transfers can generally overlap computation, and unified memory could be used. The tiling scheme should also be negotiated appropriately - this needs to be benchmarked.

sk1p, 2021-02-11:
While waiting for cupy to compile (sigh) I had a look at how we could use unified memory for our GPU support. cuSignal has support for this, but it is just [a small wrapper](https://github.com/rapidsai/cusignal/blob/803aa6ba31a93be856d2d329bc31199c63ed8204/python/cusignal/utils/arraytools.py#L57) around [the numba mapped_array function](https://numba.pydata.org/numba-doc/dev/cuda-reference/memory.html)

I still want to benchmark this, but based on this, we could implement GPU-awareness into the I/O backend:

- on initializing the backend impl class, set destination to CPU or GPU
- enabling GPU support also forces the copy mode of the mmap backend
- then we directly decode the data into the shared CPU/GPU buffer
- hopefully the driver is clever enough to transfer the data in an efficient way

If this doesn't pan out, we can think about how to integrate multiple GPU streams and manually manage overlapping of transfers and compute.

uellue, 2023-02-09:
Bumping milestone since performance impact can be large for both sparse data and GPU data, and it is growing in importance.

sk1p, 2023-08-30:
Part of the equation, including initial benchmarks for transfers: https://github.com/LiberTEM/sparseconverter/issues/20