LiberTEM/LiberTEM #953: Release QA: Loading K2IS dataset "Capture 52" fails

uellue, 2021-02-05:
## How to reproduce

Load the "Capture 52" reference dataset with a fresh install of 0.6.0rc2 on Windows 10, Python 3.8.5.

## Observed behavior

[image]

```
(libertem-releasetest) C:\Users\dwebe>libertem-server
[2021-02-05 19:40:24,711] INFO [libertem.web.server.main:147] listening on localhost:9000
[2021-02-05 19:40:25,019] INFO [libertem.web.connect.get:22] ConnectHandler.get
[2021-02-05 19:40:25,023] INFO [libertem.web.config.get:22] ConfigHandler.get
[2021-02-05 19:40:25,023] INFO [numba.cuda.cudadrv.driver.initialize:236] init
[2021-02-05 19:40:25,067] INFO [libertem.web.base.log_message:23] message: CONFIG
[2021-02-05 19:40:25,445] INFO [libertem.web.config.get:22] ConfigHandler.get
[2021-02-05 19:40:25,446] INFO [libertem.web.base.log_message:23] message: CONFIG
[2021-02-05 19:40:25,515] INFO [libertem.web.config.get:22] ConfigHandler.get
[2021-02-05 19:40:25,515] INFO [libertem.web.base.log_message:23] message: CONFIG
[2021-02-05 19:40:25,523] INFO [libertem.web.connect.get:22] ConnectHandler.get
[2021-02-05 19:40:30,216] INFO [libertem.web.base.log_message:23] message: INITIAL_STATE
distributed.worker - WARNING -  Compute Failed
Function:  _do_detect
args:      ()
kwargs:    {}
Exception: DataSetException("OSError('Unable to open file (file signature not found)')")

distributed.worker - WARNING -  Compute Failed
Function:  check_valid
args:      ()
kwargs:    {}
Exception: DataSetException('invalid magic number: 0 != 102')

[2021-02-05 19:40:33,887] INFO [libertem.web.base.log_message:23] message: DATASET_DETECTED
[2021-02-05 19:40:35,350] INFO [libertem.web.dataset.prime_numba_caches:38] starting warmup
distributed.nanny - WARNING - Restarting worker
distributed.nanny - WARNING - Restarting worker
distributed.nanny - WARNING - Restarting worker
[2021-02-05 19:40:57,103] ERROR [libertem.web.base.log_message:21] message: CREATE_DATASET_ERROR (dataset=b3f5f34c-fbe9-429f-ba5a-e95c3f3258b7)
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\web\dataset.py", line 69, in put
    await self.prime_numba_caches(ds)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\web\dataset.py", line 43, in prime_numba_caches
    await executor.run_each_host(functools.partial(prime_numba_cache, ds=ds))
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\executor\base.py", line 330, in run_each_host
    return await sync_to_async(fn_with_args, self._pool)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\executor\base.py", line 231, in sync_to_async
    return await loop.run_in_executor(pool, fn)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\concurrent\futures\thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\executor\dask.py", line 356, in run_each_host
    result_map = {
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\executor\dask.py", line 357, in <dictcomp>
    host: future.result()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\client.py", line 224, in result
    raise exc.with_traceback(tb)
distributed.scheduler.KilledWorker: ('prime_numba_cache-27f14c3b-3c2c-48a8-9d64-00ad77422dbe', <Worker 'tcp://192.168.178.84:51749', name: default-service-0, memory: 0, processing: 1>)
distributed.nanny - WARNING - Restarting worker
```

uellue, 2021-02-05:
It seems this is caused by #943, which was merged after 0.6.0rc2 was baked. Time for 0.6.0rc3, perhaps after some of the other pending PRs are finalized? :-)

sk1p, 2021-02-11:
This seems to be a Windows-specific problem, it seems to work well on Linux. Can you check which version of `distributed` you are using? I also just re-tested and confirmed it still works with the recently released distributed 2021.2.0

> It seems this is caused by #943, which was merged after 0.6.0rc2 was baked. Time for 0.6.0rc3, perhaps after some of the other pending PRs are finalized? :-)

So you can confirm that going back to v0.6.0rc2 fixes this issue? Agreed on rc3, I'll work towards this today.

sk1p, 2021-02-11:
Could you also re-run this with debug logging enabled? The log above just says 

```
distributed.nanny - WARNING - Restarting worker
```

But we don't have a clue how the worker exited. Specifically, enabling debug logging for `distributed.process` and `distributed.nanny` could give us some insights, as [described in the dask documentation](https://docs.dask.org/en/latest/debugging.html#logs). Is the problem also triggered when running the data tests on Windows? As #943 is related to an out-of-bounds situation, could you also try to run with [numba bounds checking enabled](https://numba.pydata.org/numba-doc/dev/reference/envvars.html#envvar-NUMBA_BOUNDSCHECK)? You may need to clear the numba compilation cache to make this work (and clear afterwards - the boundscheck environment variable is not included as a cache key yet)

uellue, 2021-02-11:
> So you can confirm that going back to v0.6.0rc2 fixes this issue?

No, I was testing 0.6.0rc2 and thought the bug was the one you fixed in #943. I'll check in more depth now.

sk1p, 2021-02-11:
Ah, sorry, I misunderstood, and thought that this was caused by the changes in PR #943. Then it is quite possible that this is fixed, yes.

uellue, 2021-02-11:
Fix confirmed in the current master. Sorry for the confusion, this issue was just meant as a reminder to repeat this check again with a new RC before releasing, since RC2 wasn't "clean".

sk1p, 2021-02-11:
Ah, good :+1:

uellue, 2021-02-12:
Confirmed to work in 0.6.0rc3, closing! :-)

closed 2021-02-12 by uellue