LiberTEM/LiberTEM #954: Opening a large chunked HDF5 file with gzip compression freezes LiberTEM for a long time

uellue, 2021-02-05:
Opening `reference\NeXus\MPSD/Reg5_00001.nxs` pushes the CPU load of LiberTEM to 100 % and makes it unresponsive for minutes. Closing the dataset, trying to browse or trying to shut down the cluster only lead to spinners. It eventually does shut down in the background, but the GUI stays spinning forever.

## Assumed reason

Gzip compression in HDF5 is inefficient and processing a partition takes much longer than usual for that reason.

## Expected behavior

- LiberTEM should at least shut down cleanly and quickly if workers get stuck on long-running tasks. Ideally, closing a dataset or removing an analysis shuts down the offending tasks quickly.
- The GUI should respond to the backend being shut down and not show eternal spinners, even if desperate users start mashing buttons.

## Workaround

Ctrl-C or killing the process; reloading the browser window. Avoid chunking or compression in HDF5.

sk1p, 2021-02-11:
Discussion with @uellue:

The canceling behavior is mostly inherited from `distributed`, we could improve on this by implementing #382, which makes the "unit of work" smaller in these cases. When shutting down a local cluster, we could also be a bit more forceful and kill the worker after a given timeout (need to look into dask API if this is easily implemented)

uellue, 2021-02-16:
Closed by #972

closed 2021-02-16 by uellue