LiberTEM/LiberTEM #959: GPU: blas failures

sk1p, 2021-02-11:
## Steps to reproduce:

- start a LiberTEM server on a computer with an NVIDIA GPU
- create a local cluster with a CUDA worker
- load a data set (I used a Medipix dataset) and run a standard virtual detector analysis

## Result

```
Traceback (most recent call last):
  File "/home/alex/source/LiberTEM/src/libertem/web/jobs.py", line 47, in put
    return await self.run_udf(
  File "/home/alex/source/LiberTEM/src/libertem/web/jobs.py", line 111, in run_udf
    async for (udf_results,) in result_iter:
  File "/home/alex/source/LiberTEM/src/libertem/udf/base.py", line 1232, in run_for_dataset_async
    async for part_results, task in executor.run_tasks(tasks, cancel_id):
  File "/home/alex/source/LiberTEM/src/libertem/executor/base.py", line 294, in run_tasks
    async for i in agen:
  File "/home/alex/source/LiberTEM/src/libertem/executor/base.py", line 244, in async_generator
    item = await loop.run_in_executor(pool, inner_next, gen)
  File "/home/alex/source/pyenv/versions/3.8.6/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/alex/source/LiberTEM/src/libertem/executor/base.py", line 237, in inner_next
    return next(gen)
  File "/home/alex/source/LiberTEM/src/libertem/executor/dask.py", line 258, in run_tasks
    for future, result_wrap in as_completed:
  File "/home/alex/.virtualenvs/libertem/lib/python3.8/site-packages/distributed/client.py", line 4320, in __next__
    return self._get_and_raise()
  File "/home/alex/.virtualenvs/libertem/lib/python3.8/site-packages/distributed/client.py", line 4311, in _get_and_raise
    raise exc.with_traceback(tb)
  File "/home/alex/source/LiberTEM/src/libertem/executor/dask.py", line 98, in __call__
    "task_result": self.task(),
  File "/home/alex/source/LiberTEM/src/libertem/udf/base.py", line 918, in __call__
    return UDFRunner(self._udfs).run_for_partition(
  File "/home/alex/source/LiberTEM/src/libertem/udf/base.py", line 1151, in run_for_partition
    self._run_udfs(numpy_udfs, cupy_udfs, partition, tiling_scheme, roi, dtype)
  File "/home/alex/source/LiberTEM/src/libertem/udf/base.py", line 1083, in _run_udfs
    self._run_tile(cupy_udfs, partition, tile, device_tile)
  File "/home/alex/source/LiberTEM/src/libertem/udf/base.py", line 1049, in _run_tile
    udf.process_tile(device_tile)
  File "/home/alex/source/LiberTEM/src/libertem/udf/masks.py", line 168, in process_tile
    result = flat_data @ masks
  File "cupy/core/core.pyx", line 1033, in cupy.core.core.ndarray.__matmul__
  File "cupy/core/_routines_linalg.pyx", line 709, in cupy.core._routines_linalg.matmul
  File "cupy/core/_routines_linalg.pyx", line 427, in cupy.core._routines_linalg.dot
  File "cupy/core/_routines_linalg.pyx", line 536, in cupy.core._routines_linalg.tensordot_core
  File "cupy_backends/cuda/libs/cublas.pyx", line 964, in cupy_backends.cuda.libs.cublas.sgemmEx
  File "cupy_backends/cuda/libs/cublas.pyx", line 339, in cupy_backends.cuda.libs.cublas.check_status
cupy_backends.cuda.libs.cublas.CUBLASError: CUBLAS_STATUS_EXECUTION_FAILED
```

uellue, 2021-02-11:
It "works for me":tm: Can you reproduce this with a plain cupy `dot()`?

sk1p, 2021-02-11:
I'm now working on a minimal reproducer.

sk1p, 2021-02-11:
So, the actual problem is that support for Kepler (compute capability 3.0 aka `compute_30`) was dropped in the minor release of cuda 11.1 (in 11.0 it was still supported). I didn't find out if there is a kind-of self-test function in cupy to see if the GPU support actually works, the shortest failing example I could make up was this:

```python
cp.array(cp.zeros((1,)))
```

The `cp.array` seems to be necessary to make sure we actually wait for the exception - otherwise it is ignored and the result is `NaN`.

The exception thrown is this:

```
NVRTCError                                Traceback (most recent call last)
~/.virtualenvs/libertem/lib/python3.8/site-packages/cupy/cuda/compiler.py in compile(self, options, log_stream)
    551                     nvrtc.addAddNameExpression(self.ptr, ker)
--> 552             nvrtc.compileProgram(self.ptr, options)
    553             mapping = None

cupy_backends/cuda/libs/nvrtc.pyx in cupy_backends.cuda.libs.nvrtc.compileProgram()

cupy_backends/cuda/libs/nvrtc.pyx in cupy_backends.cuda.libs.nvrtc.compileProgram()

cupy_backends/cuda/libs/nvrtc.pyx in cupy_backends.cuda.libs.nvrtc.check_status()

NVRTCError: NVRTC_ERROR_INVALID_OPTION (5)

During handling of the above exception, another exception occurred:

CompileException                          Traceback (most recent call last)
<ipython-input-16-a74174bbc0aa> in <module>
      1 import cupy as cp
----> 2 cp.array(cp.zeros((1,)))

~/.virtualenvs/libertem/lib/python3.8/site-packages/cupy/_creation/from_data.py in array(obj, dtype, copy, order, subok, ndmin)
     39 
     40     """
---> 41     return core.array(obj, dtype, copy, order, subok, ndmin)
     42 
     43 

cupy/core/core.pyx in cupy.core.core.array()

cupy/core/core.pyx in cupy.core.core.array()

cupy/core/core.pyx in cupy.core.core.ndarray.astype()

cupy/core/core.pyx in cupy.core.core.ndarray.astype()

cupy/core/_kernel.pyx in cupy.core._kernel.ufunc.__call__()

cupy/core/_kernel.pyx in cupy.core._kernel.ufunc._get_ufunc_kernel()

cupy/core/_kernel.pyx in cupy.core._kernel._get_ufunc_kernel()

cupy/core/_kernel.pyx in cupy.core._kernel._get_simple_elementwise_kernel()

cupy/core/core.pyx in cupy.core.core.compile_with_cache()

~/.virtualenvs/libertem/lib/python3.8/site-packages/cupy/cuda/compiler.py in compile_with_cache(source, options, arch, cache_dir, extra_source, backend, enable_cooperative_groups, name_expressions, log_stream)
    371             name_expressions, log_stream, cache_in_memory)
    372     else:
--> 373         return _compile_with_cache_cuda(
    374             source, options, arch, cache_dir, extra_source, backend,
    375             enable_cooperative_groups, name_expressions, log_stream,

~/.virtualenvs/libertem/lib/python3.8/site-packages/cupy/cuda/compiler.py in _compile_with_cache_cuda(source, options, arch, cache_dir, extra_source, backend, enable_cooperative_groups, name_expressions, log_stream, cache_in_memory)
    401     if base is None:
    402         # This is checking of NVRTC compiler internal version
--> 403         base = _preprocess('', options, arch, backend)
    404         _empty_file_preprocess_cache[env] = base
    405 

~/.virtualenvs/libertem/lib/python3.8/site-packages/cupy/cuda/compiler.py in _preprocess(source, options, arch, backend)
    309         prog = _NVRTCProgram(source, '')
    310         try:
--> 311             result, _ = prog.compile(options)
    312         except CompileException as e:
    313             dump = _get_bool_env_variable(

~/.virtualenvs/libertem/lib/python3.8/site-packages/cupy/cuda/compiler.py in compile(self, options, log_stream)
    561         except nvrtc.NVRTCError:
    562             log = nvrtc.getProgramLog(self.ptr)
--> 563             raise CompileException(log, self.src, self.name, options,
    564                                    'nvrtc' if not runtime.is_hip else 'hiprtc')
    565 

CompileException: nvrtc: error: invalid value for --gpu-architecture (-arch)
```

What do you think about this patch, to make sure we don't offer GPU workers on unsupported hardware:

```
diff --git a/src/libertem/utils/devices.py b/src/libertem/utils/devices.py
index f118b1eb..3166bead 100644
--- a/src/libertem/utils/devices.py
+++ b/src/libertem/utils/devices.py
@@ -12,9 +12,10 @@ logger = logging.getLogger(__name__)
 try:
     import cupy
     cupy.cuda
+    cupy.array(cupy.zeros((1,)))
 except ModuleNotFoundError:
     cupy = None
-except (ImportError, AttributeError) as e:
+except Exception as e:
     # Cupy can be a bit fragile; allow running LiberTEM with
     # messed-up installation
     warnings.warn(repr(e), RuntimeWarning)
```

The exception handling is a bit more broad, but on the other hand, it should make sure there really is a working cupy installation. We could also move the logic to `has_cupy`, if that makes more sense.

uellue, 2021-02-11:
> What do you think about this patch, to make sure we don't offer GPU workers on unsupported hardware:

Sounds good, nice catch! :-)

sk1p, 2021-02-11:
Also reported upstream: https://github.com/cupy/cupy/issues/4660

PR #962 (2021-02-11): Catch more broken cupy installations
This includes the case of missing compute capability support in the installed CUDA runtime. Check moved to `has_cupy`.

Fixes #959

## Contributor Checklist:

* [ ] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [x] I have added/updated test cases
* [ ] I have included the [... cut]
closed 2021-02-11 by sk1p