LiberTEM/LiberTEM #963: Tcl_AsyncDelete error after pytest doctest teardown

sk1p, 2021-02-11:
Maybe matplotlib related? Or does anything else use tkinter in our codebase?

https://github.com/matplotlib/matplotlib/issues/12085 https://github.com/matplotlib/matplotlib/pull/15504

https://bugs.python.org/issue39093

https://dev.azure.com/LiberTEM/LiberTEM/_build/results?buildId=284&view=logs&j=478e0699-6756-5119-c947-4f012c6c9842&t=cae07199-0aab-54f1-f34f-6a7e15fb47de

```
2021-02-11T19:35:08.3447194Z Tcl_AsyncDelete: async handler deleted by the wrong thread
2021-02-11T19:35:08.3448143Z Windows fatal exception: code 0x80000003
2021-02-11T19:35:08.3448495Z 
2021-02-11T19:35:08.3449017Z Thread 0x00000a78 (most recent call first):
2021-02-11T19:35:08.3449664Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 300 in wait
2021-02-11T19:35:08.3450467Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\queue.py", line 179 in get
2021-02-11T19:35:08.3474330Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\threadpoolexecutor.py", line 51 in _worker
2021-02-11T19:35:08.3475809Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3476948Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3478051Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3478569Z 
2021-02-11T19:35:08.3479247Z Thread 0x0000181c (most recent call first):
2021-02-11T19:35:08.3480208Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 296 in wait
2021-02-11T19:35:08.3481226Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\queue.py", line 170 in get
2021-02-11T19:35:08.3482326Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\process.py", line 218 in _watch_message_queue
2021-02-11T19:35:08.3483430Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3484771Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3485854Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3486145Z 
2021-02-11T19:35:08.3486505Z Thread 0x00001234 (most recent call first):
2021-02-11T19:35:08.3487038Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 296 in wait
2021-02-11T19:35:08.3487613Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\queue.py", line 170 in get
2021-02-11T19:35:08.3488223Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\process.py", line 218 in _watch_message_queue
2021-02-11T19:35:08.3488831Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3489443Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3490073Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3490351Z 
2021-02-11T19:35:08.3490727Z Thread 0x00001098 (most recent call first):
2021-02-11T19:35:08.3491262Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 296 in wait
2021-02-11T19:35:08.3491841Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\queue.py", line 170 in get
2021-02-11T19:35:08.3492448Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\process.py", line 218 in _watch_message_queue
2021-02-11T19:35:08.3493060Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3493671Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3494274Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3494573Z 
2021-02-11T19:35:08.3494947Z Thread 0x000005e8 (most recent call first):
2021-02-11T19:35:08.3495581Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 300 in wait
2021-02-11T19:35:08.3496151Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\queue.py", line 179 in get
2021-02-11T19:35:08.3496762Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\threadpoolexecutor.py", line 51 in _worker
2021-02-11T19:35:08.3497365Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3497971Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3498585Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3498873Z 
2021-02-11T19:35:08.3499261Z Thread 0x00000410 (most recent call first):
2021-02-11T19:35:08.3499799Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\profile.py", line 269 in _watch
2021-02-11T19:35:08.3501172Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3501789Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3502408Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3502684Z 
2021-02-11T19:35:08.3503076Z Current thread 0x000001bc (most recent call first):
2021-02-11T19:35:08.3503658Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\asyncio\base_events.py", line 660 in call_later
2021-02-11T19:35:08.3504293Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\platform\asyncio.py", line 215 in call_at
2021-02-11T19:35:08.3504902Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\ioloop.py", line 579 in add_timeout
2021-02-11T19:35:08.3505575Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\gen.py", line 633 in with_timeout
2021-02-11T19:35:08.3506162Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\locks.py", line 252 in wait
2021-02-11T19:35:08.3506847Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\batched.py", line 78 in _background_send
2021-02-11T19:35:08.3507445Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\gen.py", line 769 in run
2021-02-11T19:35:08.3507992Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\gen.py", line 814 in inner
2021-02-11T19:35:08.3508581Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\ioloop.py", line 741 in _run_callback
2021-02-11T19:35:08.3509174Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\ioloop.py", line 688 in <lambda>
2021-02-11T19:35:08.3509767Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\asyncio\events.py", line 88 in _run
2021-02-11T19:35:08.3510361Z   File "d:\a\1\s\.tox\py37\lib\site-packages\nest_asyncio.py", line 100 in _run_once
2021-02-11T19:35:08.3510941Z   File "d:\a\1\s\.tox\py37\lib\site-packages\nest_asyncio.py", line 53 in run_forever
2021-02-11T19:35:08.3511543Z   File "d:\a\1\s\.tox\py37\lib\site-packages\tornado\platform\asyncio.py", line 199 in start
2021-02-11T19:35:08.3512149Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\utils.py", line 417 in run_loop
2021-02-11T19:35:08.3512739Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 870 in run
2021-02-11T19:35:08.3513349Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 926 in _bootstrap_inner
2021-02-11T19:35:08.3513966Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 890 in _bootstrap
2021-02-11T19:35:08.3514241Z 
2021-02-11T19:35:08.3514610Z Thread 0x00001420 (most recent call first):
2021-02-11T19:35:08.3515142Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 300 in wait
2021-02-11T19:35:08.3515798Z   File "c:\hostedtoolcache\windows\python\3.7.9\x64\lib\threading.py", line 552 in wait
2021-02-11T19:35:08.3516367Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\utils.py", line 337 in sync
2021-02-11T19:35:08.3516974Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\deploy\cluster.py", line 189 in sync
2021-02-11T19:35:08.3517585Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\deploy\cluster.py", line 110 in close
2021-02-11T19:35:08.3518210Z   File "d:\a\1\s\.tox\py37\lib\site-packages\distributed\deploy\spec.py", line 652 in close_clusters
```

sk1p, 2021-02-15:
PowerShell command to re-run doctests until failure:

```
$working = $true; while ($working) { try {pytest --doctest-modules --ignore=src/libertem/win_tweaks.py src/libertem/ -vv --durations=5 --ignore=src/libertem/analysis/fullmatch.py } catch { $working = $false; Write-Host $_; } }
```

uellue, 2022-07-28:
This issue didn't occur when I was running the PowerShell command for a while on my machine. Closing for now, we can see if it happens again once we have CI running on Windows again.

closed 2022-07-28 by uellue