LiberTEM/LiberTEM #964: Console shows errors when adjusting a ROI in the GUI; warnings about memory leak; lockup

uellue, 2021-02-12:
## How to reproduce

Open Capture_52 on Windows with a fresh 0.6.0rc3 install using Python 3.7.9, apply a ROI and adjust it many times while the computation is running.

During more detailed tests, it seems that EMPAD datasets are less affected and that the memory issue can only be triggered in standard deviation mode. I am not sure if the type or size of the ROI plays a role.

## Observed behavior

The console shows a variety of errors that seem to be related to canceling operations, for example:

<details>

```
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x000001438ACE6748>>, <Task finished coro=<Worker.heartbeat() done, defined at c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py:936> exception=TypeError("unsupported operand type(s) for -: 'float' and 'NoneType'")>)
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 949, in heartbeat
    for key in self.active_threads.values()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 950, in <dictcomp>
    if key in self.tasks
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
[2021-02-12 08:53:08,050] INFO [libertem.web.base.log_message:17] message: CANCEL_JOB_DONE (job=c2b13ba3-447c-4434-9115-8b2cf0aa3242)
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x000001438ACE6748>>, <Task finished coro=<Worker.heartbeat() done, defined at c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py:936> exception=TypeError("unsupported operand type(s) for -: 'float' and 'NoneType'")>)
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 949, in heartbeat
    for key in self.active_threads.values()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 950, in <dictcomp>
    if key in self.tasks
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
distributed.worker - ERROR - ('constrained', 'memory')
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 2659, in execute
    self.transition(ts, "memory", value=value)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 1553, in transition
    func = self._transitions[start, finish]
KeyError: ('constrained', 'memory')
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x000001438ACE6748>>, <Task finished coro=<Worker.execute() done, defined at c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py:2570> exception=KeyError(('constrained', 'memory'))>)
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 2659, in execute
    self.transition(ts, "memory", value=value)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 1553, in transition
    func = self._transitions[start, finish]
KeyError: ('constrained', 'memory')
[2021-02-12 08:53:08,214] INFO [libertem.web.base.log_message:17] message: CANCEL_JOB (job=986d4d31-b544-42a4-bbbf-ce0d7f5dda3e)
[2021-02-12 08:53:08,228] WARNING [libertem.web.jobs.delete:74] tried to remove unknown job c2b13ba3-447c-4434-9115-8b2cf0aa3242
[2021-02-12 08:53:08,228] INFO [libertem.web.base.log_message:17] message: CANCEL_JOB_FAILED (job=c2b13ba3-447c-4434-9115-8b2cf0aa3242)
[2021-02-12 08:53:08,246] INFO [libertem.web.base.log_message:17] message: JOB_STARTED (job=07005f93-cc5d-4f50-a627-f6cb7f34014a)
[2021-02-12 08:53:09,021] INFO [libertem.web.base.log_message:17] message: CANCEL_JOB_DONE (job=986d4d31-b544-42a4-bbbf-ce0d7f5dda3e)
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x00000134946C9988>>, <Task finished coro=<Worker.heartbeat() done, defined at c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py:936> exception=TypeError("unsupported operand type(s) for -: 'float' and 'NoneType'")>)
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 949, in heartbeat
    for key in self.active_threads.values()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 950, in <dictcomp>
    if key in self.tasks
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
[... 50 lines cut ...]
    return {"op": "cancelled-key", "key": ts._key}
AttributeError: 'NoneType' object has no attribute '_key'
```
</details>

At the beginning, the GUI seems to still settle on the correct result.

After some time, the console shows a long list of memory warnings:

```
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 3.02 GB -- Worker memory limit: 4.27 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 3.17 GB -- Worker memory limit: 4.27 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 3.47 GB -- Worker memory limit: 4.27 GB
distributed.worker - WARNING - Memory use is high but worker has no data to store to disk.  Perhaps some other process is leaking memory?  Process memory: 3.02 GB -- Worker memory limit: 4.27 GB
```

At that point, LiberTEM seems to be locked up. The CPU load is very low, the GUI results don't update, a new analysis added at this point never shows any display, closing the dataset or shutting down the cluster doesn't work. Even Ctrl-C doesn't end it, only closing the console window finally kills the processes.

## Expected behavior

Resizing ROIs and canceling works smoothly. In particular, it doesn't lead to memory leaks or lock-ups.

## Analysis

At first I thought it was just a cosmetic issue, but the combination of apparent memory leak and locking up points at something actually going wrong. Might be LiberTEM or an upstream bug.

sk1p, 2021-02-12:
See also #894 and #814

> At first I thought it was just a cosmetic issue, but the combination of apparent memory leak and locking up points at something actually going wrong. Might be LiberTEM or an upstream bug.

The lock up is somehow a feature of distributed - it halts processing if it detects high memory usage. Did you check that your system was actually under memory pressure, and the workers were really using that much memory? Can you retry with disabled memory limits (setting `memory_limit=0' via the `cluster_kwargs` or similar when creating the local cluster should do the trick)

I'll try to find the root cause - if it's really a leak, mem usage with memory limit disabled should be unbounded.

sk1p, 2021-02-12:
```
distributed.worker - ERROR - ('constrained', 'memory')
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 2659, in execute
    self.transition(ts, "memory", value=value)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\distributed\worker.py", line 1553, in transition
    func = self._transitions[start, finish]
KeyError: ('constrained', 'memory')
```

Hmm, there is something fishy going on with the worker states. At that place in `execute` the state should really be `executing` or similar, as a `constrained` worker transitions to `executing` before calling `execute`.

Could you try to also pass `validate=True` to the worker via `cluster_kwargs` or similar? This activates a few additional `assert`s.

uellue, 2021-02-12:
It is definitely a memory leak. The memory is lost in the worker processes according to the Task Manager.

With the additional options and running scheduler, workers and libertem-server separately, I get these messages from the workers;

<details>

```
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x0000014DE8B0B4C0>>, <Task finished name='Task-651' coro=<Worker.heartbeat() done, defined at C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\worker.py:930> exception=KeyError('<TaskProxy: <UDFTask [<libertem.udf.stddev.StdDevU-5fa05823bf89942089de25e53270690f')>)
Traceback (most recent call last):
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\worker.py", line 936, in heartbeat
    response = await retry_operation(
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\utils_comm.py", line 384, in retry_operation
    return await retry(
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\utils_comm.py", line 369, in retry
    return await coro()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\core.py", line 858, in send_recv_from_rpc
    result = await send_recv(comm=comm, op=key, **kwargs)
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\core.py", line 657, in send_recv
    raise exc.with_traceback(tb)
  File "c:\users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\core.py", line 493, in handle_comm
    result = handler(comm, **msg)
  File "c:\users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\scheduler.py", line 2196, in heartbeat_worker
    ws._executing = {
  File "c:\users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\scheduler.py", line 2197, in <dictcomp>
    self.tasks[key]: duration for key, duration in executing.items()
KeyError: '<TaskProxy: <UDFTask [<libertem.udf.stddev.StdDevU-5fa05823bf89942089de25e53270690f'
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x000002003324B4C0>>, <Task finished name='Task-678' coro=<Worker.heartbeat() done, defined at C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\worker.py:930> exception=KeyError('<TaskProxy: <UDFTask [<libertem.udf.stddev.StdDevU-83d0e91bdfe7f1901c806d22a901ace1')>)
Traceback (most recent call last):
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\worker.py", line 936, in heartbeat
    response = await retry_operation(
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\utils_comm.py", line 384, in retry_operation
    return await retry(
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\utils_comm.py", line 369, in retry
    return await coro()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\core.py", line 858, in send_recv_from_rpc
    result = await send_recv(comm=comm, op=key, **kwargs)
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\core.py", line 657, in send_recv
    raise exc.with_traceback(tb)
  File "c:\users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\core.py", line 493, in handle_comm
    result = handler(comm, **msg)
  File "c:\users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\scheduler.py", line 2196, in heartbeat_worker
    ws._executing = {
  File "c:\users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\scheduler.py", line 2197, in <dictcomp>
    self.tasks[key]: duration for key, duration in executing.items()
KeyError: '<TaskProxy: <UDFTask [<libertem.udf.stddev.StdDevU-83d0e91bdfe7f1901c806d22a901ace1'
tornado.application - ERROR - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOLoop object at 0x0000014DE8B0B4C0>>, <Task finished name='Task-722' coro=<Worker.heartbeat() done, defined at C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\worker.py:930> exception=KeyError('<TaskProxy: <UDFTask [<libertem.udf.stddev.StdDevU-a8b30dcfebcfe6c17a6c082c0557ffd6')>)
Traceback (most recent call last):
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\tornado\ioloop.py", line 741, in _run_callback
    ret = callback()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\tornado\ioloop.py", line 765, in _discard_future_result
    future.result()
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\worker.py", line 936, in heartbeat
    response = await retry_operation(
  File "C:\Users\dwebe\.conda\envs\libertem\lib\site-packages\distributed\utils_comm.py", line 384, in retry_operation
    return await retry(
[... cut ...]
Alternating between StdDev and Average, waiting for it to finish. There is a slight increase in total memory use, but it is not a leak since the memory consumed by the worker processes remains constant.

[image]

Resizing ROI with Average: Increasing significantly, but not as fast as StdDev.

[image]

## My Conclusion

The memory is lost in the workers when tasks are cancelled. It is not a memory leak in the UDFs or the reader during normal execution since these can be executed repeately without increasing memory consumption.

uellue, 2021-02-12:
Is this a show-stopper for the 0.6 release? IMO LiberTEM can still be used successfully with this bug and the workaround is simple: Don't play with ROIs excessively, and restart in case memory consumption gets out of hand. Furthermore, it looks like an issue that may take some time to track down and fix. In that sense I'd tend to leave this for 0.7 or 0.6.1. What is your take?

uellue, 2021-02-12:
The memory consumption doesn't show properly in the task manager.

## Tasks right after using

The Command Processor with most memory use runs the workers.

[image]

## Tasks after longer settling period

The Command Processor with the workers significantly reduced its displayed memory consumption.

[image]

## Total memory use for previous view

[image]

## Total memory use after killing the Command Processor with all workers

A drop of 12 GB in memory consumption

[image]

## Tasks after killing the Command Processor with the workers

[image]

## Performance Monitor of a worker process

Monitoring memory-related values for a worker process while triggering the bug. "Working Set" seems to be the culprit. The amount of slightly less than 4 GB checks out since there are four such workers. The scale is a.u., i.e. the values are not comparable between curves.

[image]

## Conclusion

The memory is lost in an odd way -- where exactly may give a hint on what is going on here.

uellue, 2021-02-12:
@sk1p can you trigger the behavior on Linux, by the way?

sk1p, 2021-02-12:
> It is definitely a memory leak

If you re-trigger the problem indefinitely, do you hit operating system limits,. i.e. allocation erros?

> can you trigger the behavior on Linux, by the way?

I didn't have time to reproduce this yet, but I remember having hit some of these `distributed` messages before. It's possible that somehow canceled results aren't properly released, which is a Problem with K2-sized frames. I'll try to create an automated stress-test for this case.

sk1p, 2021-02-15:
I've reproduced the issue on Linux and created a minimal reproducer just using `distributed`: https://github.com/dask/distributed/issues/4510. This triggers a memory leak, and also the `KeyError` related to the heartbeat functionality.

I couldn't reliably trigger the transition-related error (only manually in LiberTEM) - can you try to run the reproducer to see if it triggers on your system?

sk1p, 2021-02-15:
Could you also try the following patch:

```diff
diff --git a/src/libertem/executor/dask.py b/src/libertem/executor/dask.py
index 318250ef..398d6ac8 100644
--- a/src/libertem/executor/dask.py
+++ b/src/libertem/executor/dask.py
@@ -2,6 +2,7 @@ from copy import deepcopy
 import functools
 import logging
 import signal
+import gc
 
 from dask import distributed as dd
 
@@ -270,6 +271,7 @@ class DaskJobExecutor(CommonDaskMixin, JobExecutor):
         if cancel_id in self._futures:
             futures = self._futures[cancel_id]
             self.client.cancel(futures)
+            self.client.run(gc.collect)
 
     def run_each_partition(self, partitions, fn, all_nodes=False):
         """
```

On my system, it takes a very long time and many repeats of manual action to reproduce this issue in LiberTEM, and this patch doesn't seem to fix the issue. As it also appears to be an issue when only using dask distributed, if this patch doesn't fix it, I would suggest we postpone this issue for 0.7. Agreed?

uellue, 2021-02-16:
> I couldn't reliably trigger the transition-related error (only manually in LiberTEM) - can you try to run the reproducer to see if it triggers on your system?

On my system I also only get the `KeyError` with the reproducer.

uellue, 2021-02-16:
When I run the example adding the `gc.collect` part, my system runs out of memory within seconds.

uellue, 2021-02-16:
> Could you also try the following patch:

The patch doesn't change the behavior in LiberTEM. Agreed to postpone this to 0.7.

As a mitigation measure, one could change the behavior of the ROI selector. Instead of recalculating for every movement, one could kick off a calculation only when the selector is dropped. That should reduce the number of started and cancelled jobs.

uellue, 2021-02-16:
Confirmed again: While garbage collection helps for https://github.com/dask/distributed/issues/4510, it does not fix the issue in LiberTEM. Could it possibly be related to memory allocated in a way where the garbage collector is not effective? Some possibilities that I could imagine:

* Allocated by Numba in or for compiled code
* Loading DLLs
* Operating system resources / buffers
* Shared memory

It seems that whatever resource remains dangling would have been released if the future ran to completion.

It would be interesting to see if the reproducer "works" with GC activated when being used to run actual LiberTEM futures.

sk1p, 2021-02-16:
I'll need to do a deep dive into this after the 0.6 release - in the reproducer, I noticed that the event loop somehow keeps a (weak?) reference to the asyncio future, which is then broken up by the garbage collector. Maybe in our case we have another reference going on somehow. Then of course the issues you mention. On Linux, it should be easy to get more detailed information on these, for example via `/proc/`.

[objgraph](https://mg.pov.lt/objgraph/) comes in handy for debugging these kinds of problems.

PR #1119 (2021-09-30): Work around memory leak, fixes #894
The root cause for #894 is (unbounded) caching of functions that are submitted to the dask cluster - we need to make sure we use a consistent function and not directly submit callables that are generated on the fly.

Closes #964, closes #894

## Contributor Checklist:

* [N/A] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [N/A] I have added [... cut]
closed 2021-09-30 by uellue