LiberTEM/LiberTEM #965: File opening issues Linux cluster Windows server

uellue, 2021-02-15:
During testing 0.6.0rc3, item "Run libertem-server on Windows, connect to a remote dask cluster running on Linux, open all file types and perform an analysis for each file type."

## EMPAD

Detection fails. Setting manually to EMPAD: 

```
[2021-02-15 18:42:17,050] ERROR [libertem.web.base.log_message:21] message: CREATE_DATASET_ERROR (dataset=2d08dd96-7c85-44b9-bbb4-cc6704911a84)
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\web\dataset.py", line 67, in put
    ds = await load(filetype=cls, executor=executor, enable_async=True, **dataset_params)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\__init__.py", line 63, in _init_async
    ds = await run_blocking(ds.initialize, executor=executor.ensure_sync())
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\utils\async_utils.py", line 18, in run_blocking
    return await tornado.ioloop.IOLoop.current().run_in_executor(None, partial(fn, *args, **kwargs))
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\concurrent\futures\thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\empad.py", line 173, in initialize
    os.stat(self._path_raw).st_size / (
FileNotFoundError: [WinError 3] The system cannot find the path specified: '/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/EMPAD/BiFeO3/scan_11_x256_y256.raw'
```

## FRMS6

When running an analysis (in this case sum and Radial Fourier):

```
[2021-02-15 18:50:17,513] ERROR [libertem.web.jobs.put:61] error running job, params={'roi': {}}
Traceback (most recent call last):
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\web\jobs.py", line 53, in put
    details=analysis_details,
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\web\jobs.py", line 107, in run_udf
    corrections = dataset.get_correction_data()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\frms6.py", line 678, in get_correction_data
    dark=self._get_dark_frame(),
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\frms6.py", line 585, in _get_dark_frame
    for tile in tiles:
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\base\partition.py", line 281, in get_tiles
    corrections=self._corrections,
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\base\backend_buffered.py", line 247, in get_tiles
    with fileset:
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\base\fileset.py", line 65, in __enter__
    f.open()
  File "c:\users\dwebe\.conda\envs\libertem-releasetest\lib\site-packages\libertem\io\dataset\base\backend.py", line 139, in open
    f = open(self._path, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/storage/holo/clausen/testdata/ER-C-1/groups/data_science/data/reference/PNDetector/181025_008_000.frms6'
```

## Analysis

This seems to be related to running functions on the executor vs. running locally. I'll test next if the EMPAD one is a Windows vs. Linux issue.

sk1p, 2021-02-15:
EMPAD bug is related to `os.stat` not being run via the `executor. Same for FRMS6, where `_get_dark_frame` needs to run in the `executor`.

uellue, 2021-02-15:
Thx! I'll work on a fix, that should be easy enough. :-)

sk1p, 2021-02-15:
FRMS6 fix could be a tiny bit tricky - dark frame computation needs either to run eagerly while initializing (if it's enabled), or we need to keep a reference to the executor to be used when loading the dark frame.

uellue, 2021-02-15:
The FRMS6 reader is already reading the corrections in _do_initialize(), one just has to actually use these values instead of reading them again. 🙈 

By the way, we now have dataset tests and distributed tests. Would it be possible to run the dataset tests in a distributed fashion to catch such errors?

sk1p, 2021-02-15:
>The FRMS6 reader is already reading the corrections in _do_initialize(), one just has to actually use these values instead of reading them again. see_no_evil

:+1: :+1: :+1: good catch.

> By the way, we now have dataset tests and distributed tests. Would it be possible to run the dataset tests in a distributed fashion to catch such errors?

Not yet, as that would require a docker-in-docker setup, which is equivalent to root on the host system. I would not currently be comfortable with this - it would require some more hardening in our azure agent setup.

uellue, 2021-02-15:
> Not yet, as that would require a docker-in-docker setup, which is equivalent to root on the host system. I would not currently be comfortable with this - it would require some more hardening in our azure agent setup.

Ok, nevermind then. Manual QA it is, then. :-)

PR #966 (2021-02-15): File access only through executor
Closes #965

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [ ] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [ ] I have added/updated documentation for all user-facing changes
* [ ] I have added/updated test cases

## Reviewer Checklist:

* [x] `/azp run libertem.libertem-data` passed
commit 0f72cb13 (2021-02-15): File access only through executor

Closes #965
closed 2021-02-15 by uellue