LiberTEM/LiberTEM #985: Release 0.6.1

uellue, 2021-03-01:
This is a point release, mostly for #984 since it fixes a severe performance issue with a localized change.

uellue, 2021-03-01:
A potential candidate for inclusion would be #982. However, that PR has a cross relation to the job interface removal and is not so urgent for real-world usability. I'd propose to leave that one for 0.7 to keep things simple.

uellue, 2021-03-09:
We'll wait with the release for 1-2 weeks after #984 is merged since there were more changes necessary than anticipated and we'd like to see if any issues crop up when using LiberTEM.

uellue, 2021-03-18:
@sk1p and I decided to do a release 0.7 with some of the recently merged new features since they will be useful for work in https://github.com/Ptychography-4-0/ptychography/pull/35/ and https://github.com/LiberTEM/LiberTEM-live

Scope TBD! @sk1p it should probably also include the live plotting parts with effects on the Analysis interface, as far as they affect LiberTEM and not LiberTEM-live? :-)

sk1p, 2021-03-23:
> it should probably also include the live plotting parts with effects on the Analysis interface, as far as they affect LiberTEM and not LiberTEM-live? :-)

Agreed - I've added a TODO list to the top of this issue to keep track of the "release critical" issues.

sk1p, 2021-05-31:
https://github.com/nbQA-dev/nbQA

https://github.com/computationalmodelling/nbval

PR #1047 (2021-06-10): Post-release changelog update
Fixes #985

* Date and DOI
* Update release checklist drive-by

## Contributor Checklist:

* [x] I have added or updated my entry in [the creators.json file](https://github.com/LiberTEM/LiberTEM/blob/master/packaging/creators.json)
* [x] I have added [a changelog entry](https://github.com/LiberTEM/LiberTEM/tree/master/docs/source/changelog) for my contribution
* [x] I have added/updated documentation for all user-facing changes
* [N/A] I have added/updated test cases
* [N/A] I have included the [rebuilt production build of the client](https://libertem.github.io/Li [... cut]
closed 2021-06-10 by sk1p