LiberTEM/LiberTEM #1000: Difference between self.results.buffername and self.results['buffername']

uellue, 2021-03-22:
`UDFData` implements both `__getitem__()` and `__getattr__()`. The former returns a `BufferWrapper` object, the latter a view into the memory of the underlying data array. 

This can be a bit confusing:

* When merging results in an UDF's `process_*()`method: Use `self.results.buffername` to get the proper view.
* When inside a UDF's `merge()` function: Use sth like `dest['buffername'][:] = dest['buffername'] + src['buffername']`´since the buffers are passed in as `Dict`.
* When in `get_results()`: Use `self.results['buffername']` to pass through a `BufferWrapper`, but use `self.result('buffername', some_operation(self.results.buffername))` for a final postprocessing step.

sk1p, 2021-03-22:
Related: #631

sk1p, 2021-03-22:
How about we fix this together with #1001?

> When merging results in an UDF's `process_*()` method: Use `self.results.buffername` to get the proper view

I think this is the okay and can stay like this.

> When inside a UDF's `merge()` function: Use sth like `dest['buffername'][:] = dest['buffername'] + src['buffername']` since the buffers are passed in as Dict.

Actually a `MappingProxyType`, which is slightly different, and used for making the dict readonly. This could be changed to a different object that also allows attribute access (to fix #631), which then should return the view, as it should

> When in `get_results()`: Use `self.results['buffername']` to pass through a `BufferWrapper`, but use `self.result('buffername', some_operation(self.results.buffername))` for a final postprocessing step. Alternatively, use `self.results['buffername'].raw_data` instead of `self.results.buffername`

Hmm. This makes me think we may want to remove the `__getitem__` way of getting a `ResultBuffer`, and make that more explicit? Letting `self.results.get_buffer('buffername')`  return the `ResultBuffer`, and `self.results.buffername` return the current view, as usual? Maybe you also have a better idea to get at the raw data of a buffer?

PR #1003 (2021-03-25): UDF: Consistency fixes and `get_results` improvements
Now, all accesses to buffers in `UDF.results` and the `src`/`dest` parameters of `UDF.merge` should be done via `__getattr__`; if the `BufferWrapper` is really needed, then `UDF.results.get_buffer` should be used.

As a follow-up to #994, `get_results` now should consistently return numpy arrays, which are then converted to `BufferWrapper`s before returning them to the user. This means we can make `get_results` a bit more DRY by not having to repeat buffer names. Also, buffers that don't have to be post-processed, can be left out in `get_re [... cut]
commit 71fc089e (2021-03-25): UDF: Consistency fixes and `get_results` improvements

Now, all accesses to buffers in `UDF.results` and the `src`/`dest`
parameters of `UDF.merge` should be done via `__getattr__`; if the
`BufferWrapper` is really needed, then `UDF.results.get_buffer` should
be used.

As a follow-up to #994, `get_results` now should consistently return
numpy arrays, which are then converted to `BufferWrapper`s before
returning them to the user. This means we can make `get_results` a bit
more DRY by not having to repeat buffer names. Also, buffers that don't
have to be post-processed, can be left out in `get_r [... cut]
closed 2021-03-25 by uellue