LiberTEM/LiberTEM #1001: Boilerplate in `get_results()` with many derived buffers

uellue, 2021-03-22:
Right now, a real-world `get_results()` function can look like this:

```python
...
return {
    'num_frames': self.results['num_frames'],
    'varsum': self.results['varsum'],
    'sum': self.results['sum'],
    'var': self.result(name='var', data=var),
    'std': self.result(name='std', data=np.sqrt(var)),
    'mean': self.result(name='mean', data=self.results['sum'].raw_data / num_frames),
}
```

or like this:

```python
def get_results(self):
    results = get_results(self.results)
    return {
        'fourier': self.results['fourier'],
        'complex': self.result('complex', results),
        'amplitude': self.result('amplitude', np.abs(results)),
        'phase': self.result('phase', np.angle(results)),
    }
```

The buffer name has to be repeated for each buffer. This could be improved by returning a dict of `buffername --> ndarray` and doing the conversion to a `BufferWrapper` automatically. That has two advantages: First, it saves some tedious typing, including the call to `self.result()`. Second, it enforces that results of an UDF are always `BufferWrapper`.

One could even just pass through buffers that are not in the result dict, i.e. `num_frames`, `varsum` and `sum` in the stddev example, so that one just has to worry about buffers that are to be modified or created.

sk1p, 2021-03-22:
> The buffer name has to be repeated for each buffer. This could be improved by returning a dict of buffername --> ndarray and doing the conversion to a BufferWrapper automatically. That has two [...]

Thanks, this kind of feedback is really useful. I agree that the repeated names may not be DRY enough. From the perspective of writing a `get_results` function, returning a numpy array directly would indeed be less typing and a bit more DRY.

A (maybe minor?) point is that, right now, the conversion from array to `BufferWrapper` at that point is explicit, and what you return from `get_results` is also what you get back from `run_udf`. But this may be acceptable, in combination with the easier-to-implement `get_results`? Let me know what you think.

sk1p, 2021-03-22:
> One could even just pass through buffers that are not in the result dict, i.e. `num_frames`, `varsum` and `sum` in the stddev example, so that one just has to worry about buffers that are to be modified or created.

Possibly, that also means it's no longer possible to remove buffers from the final result of the UDF. Removing results from the final result may be important to hide implementation details that could change in future versions of a UDF. Maybe this should be already declared in `get_result_buffers`? `self.buffer(..., private=True)`? What do you think? Important? Feature creep?

uellue, 2021-03-23:
> Possibly, that also means it's no longer possible to remove buffers from the final result of the UDF. Removing results from the final result may be important to hide implementation details that [...]

Since `get_results()` is already an advanced feature that not all UDFs need and that is more of , probably it would make sense to require passing through all buffers, and dropping buffers from a result could be done here? Or do we need some sort of guarantee that whatever is in `get_resul_buffers()` also is present in `get_results()`?

sk1p, 2021-03-24:
> Since `get_results()` is already an advanced feature that not all UDFs need and that is more of , probably it would make sense to require passing through all buffers, and dropping buffers from a result could be done here?

> Or do we need some sort of guarantee that whatever is in `get_resul_buffers()` also is present in `get_results()`?

I think it would be good to at least know beforehand which "result channels" will exist - the idea of `private=True` was to declare that this buffer is only used as an intermediate buffer, basically the symmetric case to `allocate=False`.

To summarize my proposal:

- All buffers are declared in `get_result_buffers`
- If a buffer is only computed in `get_results`, it should be marked via `allocate=False` so it isn't allocated on workers
- If a buffer is only used as intermediary result, it should be marked via `private=True`
- Not including a buffer in `get_results` means it will either be passed on unchanged, or dropped if `private=True`
- It's an error to omit an `allocate=False` buffer in `get_results`
- It's an error to include a `private=True` buffer in `get_results`

So, your example from above would become:

```python
return {
    'var': var,
    'std': np.sqrt(var),
    'mean': self.results['sum'].raw_data / num_frames,
}
```

`num_frames`, `varsum`, `sum` aren't declared private in this case and returned unmodified. As an example, if one would like to not include `varsum` in the final result, one could declare it private in `get_result_buffers`, then it would be dropped and visualization would skip it.

With the proposed change from #1000, it would become:

```python
return {
    'var': var,
    'std': np.sqrt(var),

    # explicitly get result buffer, to mitigate confusion of __getitem__ vs __getattr__
    'mean': self.results.get_buffer('sum').raw_data / num_frames,
}
```

Thoughts? Maybe `private` and `allocate` can be combined into a single parameter, then we also automatically exclude the invalid case `private == True and allocate == False`. Maybe `use='private'` and `use='result_only'` (to be bike-shedded, please!)

PR #1003 (2021-03-25): UDF: Consistency fixes and `get_results` improvements
Now, all accesses to buffers in `UDF.results` and the `src`/`dest` parameters of `UDF.merge` should be done via `__getattr__`; if the `BufferWrapper` is really needed, then `UDF.results.get_buffer` should be used.

As a follow-up to #994, `get_results` now should consistently return numpy arrays, which are then converted to `BufferWrapper`s before returning them to the user. This means we can make `get_results` a bit more DRY by not having to repeat buffer names. Also, buffers that don't have to be post-processed, can be left out in `get_re [... cut]
closed 2021-03-25 by uellue