LiberTEM/LiberTEM #1042: Clustering analysis usability: cancellation, memory usage, parallelization?

sk1p, 2021-06-09:
The clustering analysis in the GUI can be a bit user-unfriendly. Depending on parameters, it can take a long time (it's not parallelized yet!), and also a lot of memory (`O(n^2)` IIRC). 

- [ ] the user should be able to cancel the clustering analysis (currently, the `AgglomerativeClustering().fit()` call keeps on running when removing an analysis in the GUI)
- [ ] the server should shutdown gracefully while a clustering analysis is running (currently it doesn't: ctrl+c doesn't work properly and I had to kill the process)

sk1p, 2021-09-27:
@uellue I don't think this issue can be solved as part of 0.8.

> Depending on parameters, it can take a long time (it's not parallelized yet!), and also a lot of memory (O(n^2) IIRC).

IIRC this is a property of the algorithm that is used, so to fix it, we would need to select another clustering algorithm

> the `AgglomerativeClustering().fit()` call keeps on running when removing an analysis in the GUI

This is an issue with the implementation of the clustering algorithm, i.e. sklearn. The `fit` function is (inherently?) synchronous, so there is no way to cancel a running computation. A (stupid) work-around would be to run the `fit` in a subprocess, i.e. in its own Python interpreter, which we could `kill` on cancellation. This would still not fix the memory problem, or rather create additional memory issues

> the server should shutdown gracefully while a clustering analysis is running

Maybe this part of the issue can be fixed by running the synchronous part (i.e. `fit`) of the analysis on a `daemon=True` thread? I think this is the easiest part and possible to fix for 0.8, let me know if I should look into it, or if you want to give it a go, @uellue.

uellue, 2021-10-01:
I'm putting this in the backlog for now since I am not aware of anyone having a real-world issue with this. :-)