LiberTEM/LiberTEM #1045: Fast turn-around of CI pipeline

uellue, 2021-06-09:
We are supporting quite a number of Python versions now (good) and have a lot of tests (good). As a downside, CI takes a long time to complete. We should consider options to speed things up:

* Throw hardware at the problem and run tests in parallel where possible. See also #683 
* Use several CI services in parallel
* Compartmentalize the code and tests to only run tests that are relevant for a specific change. For example, continue splitting LiberTEM in sub-modules #261 
* Review tests: Reduce time for long-running tests; eliminate unnecessary tests. See also discussion in #914 
* Possibly only run the full test matrix for release candidates

sk1p, 2021-06-09:
Using pre-commit.ci for flake8, mypy etc. #1036 should also help, as we don't "waste" azure pipeline jobs for these things.

sk1p, 2021-06-10:
Identifying slow tests, take from the currently slowest run (example: py36 on Windows):

`libertem` main test suite:

```
============================ slowest 10 durations =============================
27.72s call     tests/corrections/test_corrset.py::test_patch_corr_odd
17.04s call     tests/test_analysis_clust.py::test_cluster_analysis
13.93s call     tests/server/test_copy.py::test_copy_notebook
13.72s call     tests/corrections/test_detector.py::test_detector_patch_large
12.01s call     tests/template/test_clust_template.py::test_clust_default
11.98s call     tests/io/test_cached.py::test_with_roi
10.42s call     tests/test_analysis_raw.py::test_pick_analysis
9.75s call     tests/udf/test_stddev.py::test_stability
9.60s call     tests/common/test_numba.py::test_numba_prime
9.43s call     tests/template/test_sd_template.py::test_sd_roi
= 710 passed, 231 skipped, 21 deselected, 4 xfailed, 26 warnings in 619.45s (0:10:19) =
```

`nbval`:

```
============================= slowest 10 durations =============================
73.37s call     examples/radialfourier.ipynb::Cell 8
34.02s call     examples/pcmclustering.ipynb::Cell 19
10.80s call     examples/Introduction to UDFs.ipynb::Cell 7
8.09s call     examples/Introduction to UDFs.ipynb::Cell 25
6.96s call     examples/simple libertem API.ipynb::Cell 5
6.24s call     examples/pcmclustering.ipynb::Cell 11
5.96s call     examples/pcmclustering.ipynb::Cell 8
5.00s call     examples/pcmclustering.ipynb::Cell 2
4.92s call     examples/pcmclustering.ipynb::Cell 17
3.99s call     examples/pcmclustering.ipynb::Cell 5
======================= 200 passed in 263.17s (0:04:23) ========================
```

We should definitely "fix" the radialfourier notebook - do you think we can reduce the number of bins or orders while still having a good example?

If we have a concrete user of the clustering code, we could also [work on improving the algorithm used](https://scholar.google.com/scholar?hl=en&as_sdt=0%2C5&q=Efficient+Parallel+Hierarchical+Clustering&btnG=)

`libertem-data`:

```
============================= slowest 10 durations =============================
27.82s call     tests/io/datasets/test_k2is.py::test_comparison
22.60s call     tests/io/datasets/test_blo.py::test_correction[False]
18.56s call     tests/io/datasets/test_ser.py::test_correction[False]
18.20s call     tests/io/datasets/test_k2is.py::test_comparison_roi
18.09s call     tests/io/datasets/test_mib.py::test_correction[False]
17.01s call     tests/io/datasets/test_k2is.py::test_comparison_mask
16.53s setup    tests/io/datasets/test_frms6.py::test_comparison
12.27s call     tests/io/datasets/test_blo.py::test_correction[True]
11.89s call     tests/io/datasets/test_dm.py::test_correction[False]
11.58s call     tests/io/datasets/test_k2is.py::test_read
==== 369 passed, 1 skipped, 15 deselected, 6 warnings in 461.98s (0:07:41) =====
```

> Possibly only run the full test matrix for release candidates

As a note, we do have `functional` and `slow` test markers, but didn't keep them up to date. We could:

a) Re-add `slow` markers for all tests that take more than a threshold to run (3s? or even 1s?)
b) Add a manually triggered pipeline (or the GHA equivalent) to run all tests marked as slow on all combinations (OS × Python Version)

How do we make sure that no new slow tests are introduced? We could add a "review tests and mark slow ones" into the release checklist, and in addition use `pytest-timeout` to make sure that nothing takes an unreasonable amount of time, let's say more than 5s.

sk1p, 2021-06-22:
> Using pre-commit.ci for flake8, mypy etc. #1036 should also help, as we don't "waste" azure pipeline jobs for these things.

This is now done in #1049 - first PR with it enabled is #1050. I've decided to disable mypy for now, as it doesn't have much value with our currently limited type coverage.

uellue, 2025-05-09:
Looking at the original issue, we are now doing all of this: Data tests on own powerful CI machine, split up in submodules (in particular sparseconverter is test-intensive), mark tests as slow and only run them for RCs, retire Python versions in a timely fashion.

Currently, the run times feel "good enough" for me and quite worth it for the number of issues we catch with them.

Closing here, unless @sk1p or @matbryan52 think we should do more and better!

closed 2025-05-09 by uellue