LiberTEM/LiberTEM #1054: Unable to load HDF5 datasets saved using non-standard compression schemes (hdf5plugin)

matbryan52, 2021-06-28:
Right now a user trying to load a HDF5 file saved with non-standard compression is met with a (somewhat cryptic) error which originates with `h5py`, specifically:

```
`*** OSError: Can't read data (can't open directory: /usr/local/hdf5/lib/plugin)`
```

One method to read (and create) these non-standard compressed files is using the `hdf5plugin` package (https://pypi.org/project/hdf5plugin/) which just requires you to modify the h5py import:

```
import h5py
import hdf5plugin
```

Normally a user would do this themselves before any other import of `h5py` in order to register the plugins on the backend. However, as LiberTEM/Dask spawn separate processes this background registration is not carried through, leading to the above error at the moment a new Task is spawned even if the main node can load and read the file correctly.

A simple fix would be to modify the import in the file [io/datasets/hdf5.py](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/io/dataset/hdf5.py) as above, or perhaps better:

```
import h5py
try:
    import hdf5plugin
except ModuleNotFoundError:
    logger.debug("Unable to load hdf5plugins package, some HDF5 compression schemes will not be available")
```

as the use case is admittedly quite small and to prevent adding an extra dependency.

uellue, 2021-06-28:
Good catch! We can make this an "extra-requires" in setup.py, catch the exception and advise users to install hdf5plugins in the error message, and of course document it. Should be easy enough!

uellue, 2021-06-28:
Ah, we should also test how this behaves on Windows since "can't open directory: /usr/local/hdf5/lib/plugin" sounds pretty system-dependent...

uellue, 2021-06-28:
@matbryan52 Could you perhaps make a file available that reproduces the issue? We could add this to our test suite.

matbryan52, 2021-06-28:
> catch the exception and advise users to install hdf5plugins

So this error is (normally) raised when `read_direct` is called on the `h5py` Dataset object i.e. really quite late in execution (in fact normally on the worker nodes). What I don't know and think would be hard to test are the other situations that an `OSError` might be raised when reading part of the dataset into memory, unrelated to plugins and compression. This is is why I didn't suggest catching the error and instead just pro-actively trying the import to add the compression plugins if they are available.

> Ah, we should also test how this behaves on Windows since "can't open directory: /usr/local/hdf5/lib/plugin" sounds pretty system-dependent...

I will check what the error is under Windows.

> Could you perhaps make a file available that reproduces the issue? We could add this to our test suite.

It's a delicate one to test as we need to generate a non-standard compressed file using the plugins, and then test the case of having succeeded to import the plugins (and not). Depending on whether we want to actually try catching the above OSError we might only need to add the import and check that the ModuleNotFound catch block works. 

I will see what I can come up with for a useful test that at least verifies that we can read a file of this type.

matbryan52, 2021-06-28:
A reproducing file: https://gist.github.com/matbryan52/07a584b3de3569213731583d1ee472d4

This handles both creating the compressed hdf5 file in a temporary directory and testing loading it (in a secondary process, hence the odd structure).

uellue, 2021-06-28:
> It's a delicate one to test as we need to generate a non-standard compressed file using the plugins, and then test the case of having succeeded to import the plugins (and not).

We already had one like this before and figured out how to mock a missing or failing import for testing: https://github.com/LiberTEM/LiberTEM/blob/master/tests/test_win_tweaks.py

Essentially, one uses monkeypatching to "un-import" the module by temporarily deleting it from `sys.modules`, then injects an import function that throws the desired error for the desired module, and finally performs the action that is to be tested.

matbryan52, 2021-06-29:
There is a particularity here which is that hdf5plugin does its work in the background at import time by loading some .so/.dll files and giving them to h5py; you don't actually need to use/reference the module to gain the de-compression/reading functionality, only import it (it would show as an unused import with a linter!).

I did a bit more testing and now I think this issue only occurs when you explicitly create an external `dask-scheduler` and `dask-worker` (either local or remote workers). If you leave it to LiberTEM to create the (local) executor and cluster setup, rather than connecting to an existing cluster, then the `hdf5plugin` import is carried through from the launching process to the workers. I am wondering if this is because the main process is forked (keeping the same state) rather than the workers being fully independent python instances which run some (but clearly not all) imports from the main process.

So for most use cases the user already has what they need to work around this issue (`import h5py ; import hdf5plugin` at the top of their program), so it might just require a documentation change. It's only for the truly distributed case we'd have to try importing the module like I proposed above, or pointing the user to : https://docs.dask.org/en/latest/setup/custom-startup.html

closed 2021-07-12 by sk1p