LiberTEM/LiberTEM #1068: load raw binary file

YuLin999, 2021-07-16:
Hi,
I am trying to use liberTEM to analyze a 4d-stem dataset (256x256 scan) but I am having some trouble loading the dataset correctly. I can load the dataset in Image J with the following options:
32-bit real, width 128 pixels, height 128 pixels, offset to first image 0 bytes, number of images 65536, gap between images 1024 bytes, little-endian byte order. But right now I cannot load this dataset correctly using libertem (the visualization does not makes sense). 
Any suggestions? Thank you!

sk1p, 2021-07-19:
Hi!

> I am trying to use liberTEM to analyze a 4d-stem dataset (256x256 scan) but I am having some trouble loading the dataset correctly. I can load the dataset in Image J with the following options: [...]

You are correct, if you are using the RAW dataset format, this won't make sense, because we don't (yet) support setting a custom gap between images (frame headers / footers).

Looking at the specs you give, it sounds like EMPAD data, is that correct? If you select EMPAD instead, the data should load correctly.

Let me know if you have any questions, and thanks for the report!

YuLin999, 2021-07-19:
I just tried select EMPAD instead, looks like the data is reading correctly. Thank you!

sk1p, 2021-07-19:
> I just tried select EMPAD instead, looks like the data is reading correctly. Thank you!

Good to hear, thank you for confirming!

closed 2021-07-19 by sk1p