LiberTEM/LiberTEM #1070: JupyterHub and JupyterLab integration

uellue, 2021-07-20:
We should be able to kick off a LiberTEM GUI session from JupyterHub and JupyterLab.

- [ ] Infrastructure on LiberTEM side
- [ ] Configuration files / instructions / documentation for integration
- [ ] If applicable, upstream sth to JupyterHub / JupyterLab / other platforms where settings can be downloaded from to make set-up easy for users.

Builds on preliminary work from @sk1p : https://github.com/sk1p/libertem-jupyter-proxy

sk1p, 2021-07-26:
This almost works now, with the latest updates in the above mentioned `libertem-jupyter-proxy` package. Outside of jupyterhub, with a local notebook server, it spawns the `libertem-server` process and is usable.

But when running under jupyterhub, the service worker that is created as part of `create-react-app` springs into action. When loading the `manifest.json` file, it doesn't include credentials, but jupyterhub always needs credentials, even for "static" requests (because it doesn't do any differentiation at all between the requests!).

As we are not using the service worker anyways, I'll patch it out and try if that fixes all issues.

sk1p, 2021-07-26:
> * Configuration files / instructions / documentation for integration

For this one, I think most important is documenting how to install LiberTEM in a Python environment that is separate from the jupyterhub one, and setting the correct path in the `libertem-jupyter-proxy` config. Then, a little bit of user documentation (screnshot) on how to start from the "New" dropdown should be enough.